SmingHub/Sming #692: How to define I2C pins while using MCP23017?

ashubhatt, 2016-04-11:
Hello,

I am experimenting with MCP23017 and NodeMCU. I wonder, how to configure I2C pins for this module? 
In the given example "PortExpander_MCP23017", they do not specify pins for I2C, then which pins will the NodeMCU utilize for communication?

alonewolfx2, 2016-04-11:
Wire.pins(sda,scl);

alon24, 2016-04-11:
You mean ofcourse: void pins(int pinSCL, int pinSDA);

hreintke, 2016-04-22:
Question answered

closed 2016-04-22 by hreintke

dpechman, 2018-11-16:
Is this sample running ok if compiled with Sming 3.6.1? For me it is exploding...

Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0000
Writing 28672 @ 0x0... 28672 (100 %)
Wrote 28672 bytes at 0x0 in 2.5 seconds (90.7 kbit/s)...
Writing 196608 @ 0x7000... 196608 (100 %)
Wrote 196608 bytes at 0x7000 in 17.3 seconds (90.9 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
1 0b 26 65 e2 d5 c3 f3 73 7b 1d d1 17 96 2b ae fe 85 bc 
95 b2 3b f9 91 9a 83 b8 02 ad 98 fa d1 cf d5 d1 bb e9 32 60 ce 6c f9 e9 5d 9e cf cb 2d 70 da f4 
9d 08 39 79 f9 95 7f d1 74 92 54 77 ae 7e 92 79 5d ac ee b6 e5 7e bd 9a 2d 77 e5 66 af dd b7 68 
dc ef e2 7d 41 fc d5 cd b9 fa 59 0a ea 9a 24 05 2d ff bb ba db e5 f3 bb 0d 24 11 ce 13 39 f8 b5 
3b 7c ca a9 26 bb ed 2e 92 5e fa 9e f6 ff f6 41 21 ed dd 27 ee ae 97 5f ea b7 b6 88 10 7a 15 cf 
0f 00 32 00 fc 43 a4 1f 2d 02 10 95 39 80 bf 87 72 2f 7d 9c cc 57 79 36 07 30 6b d5 31 2a 80 03 
ac a4 1d d6 49 87 cd f9 cf ec da 06 b0 fa 53 ab 70 1d a7 05 ad c9 b1 b9 43 a4 c9 85 b6 71 ca 19 
ce 1d dd 96 8b d4 e4 90 8e 55 7b 96 6e a4 71 5b 68 e4 dc 2f 09 ec 9f d0 6f a9 52 f4 75 e6 57 93 
ca 54 d3 ad 25 91 f3 1f d9 ea 8d 7a 8c 12 b8 1c b1 b7 d3 4e 5c eb e5 11 76 37 87 49 98 24 8f 9e 
a9 41 73 26 d8 90 cc 09 d1 c9 e9 09 d2 51 04 6f 88 4a 08 e0 35 91 c3 69 fa 8b 78 27 e7 83 eb a1 
78 6e ac d2 db 14 60 04 6a 9d b6 81 3f 20 95 e9 e8 a9 ae 09 9f 8a d5 8a 53 2e 9b b6 71 cc 49 36 
1f ac 58 88 a6 4a da 65 7f de 05 87 31 a1 3d 0f 21 22 99 77 73 62 38 59 3c dd ef 57 ea e8 bb 37 
a9 3b c0 19 ff 62 a5 56 34 28 5c c7 ad 48 6e 23 a5 e7 2d 29 24 5d b9 cc 93 32 33 98 94 61 8e 55 
0f 00 33 00 fc f6 2a 3c 12 67 22 32 69 49 9c 03 11 0d 1a e0 a2 4c d6 3a 34 ac 0b 57 29 92 86 da 
8b 74 46 ad 8b 19 d5 0f 57 30 ed 2f 6c ef 5e ac ee 68 ad 5e a8 39 a8 d0 dd fa f4 54 5f b8 10 92 
b5 9a 20 88 a4 75 09 98 aa aa ea 1a 66 b0 9a 3a 7b 45 4b 1a 9a 6c c5 6a 71 0c 35 fe 9a 1a 6f 40 
c6 f4 10 98 a5 1a de 48 65 44 00 55 d0 5f 18 6f 88 4a 93 e9 80 bb a0 aa 12 1d 98 7c c3 6c 37 cf 
88 6c 3b 74 14 6c 7d a3 56 e9 8f ed 3f 27 73 9f 9e 06 bb 11 1e 31 6e 27 81 77 fe 33 cb 96 f3 75 
fe cf 57 2e 90 33 8f 2f dc d1 07 16 ae 2c 01 fe e2 94 11 3a 82 ac 19 00 a8 45 be e7 a8 63 56 11 
92 22 f7 93 ee 53 e4 9a 8e 21 9f b4 71 05 c8 5a 30 fa e9 22 ec de 98 ba 77 93 2e 88 c9 51 48 65 
8a 80 e6 9b ae 59 ff ec 89 21 33 cb f7 65 ea a7 40 08 95 65 7c d3 b5 ab 98 da 0a 67 79 2e e7 39 
0f 00 34 00 fc 98 47 4f 9f 46 92 0e b7 e5 ee f3 ca 36 ab 64 0c 9f 19 ff 9d ca b2 39 3b 53 c6 d8 
84 a5 2e 57 6e 65 c6 bd 09 b5 67 e1 d4 0a 73 6a 5a b6 19 11 0b 82 e2 e6 da bc b0 56 96 62 ea 1d 
b2 f4 4d 53 08 fc 74 25 8c c8 c2 bb f1 60 f5 c0 2b e8 0f 67 b1 99 52 2b 21 ac 3a 58 1c bc 52 69 
1d f7 aa 9a c0 19 6b 1f a4 46 f3 09 e5 09 18 91 ac ce 88 f1 94 35 5d 0e 51 65 6a 3b e0 46 c7 e0 
0f 68 7c e7 
0f 00 40 00 fc ec cc 0f 86 75 a8 ca 88 28 e5 a5 36 49 69 d6 a7 70 a3 c6 ec c7 1a e6 cf e6 3c 35 
38 7d 34 ca 08 90 54 13 84 d1 30 86 31 83 f2 71 15 63 28 9b e5 ed 09 f2 94 48 85 47 2a 5a 06 15 
31 a7 ca 3b 70 ea 72 e9 70 8b 54 24 87 1f fc 9b bc cc 08 44 13 26 0a d9 8c 79 ad da 32 b1 02 3a 
c5 1d 11 42 e3 55 dc c0 49 1a 3a 12 a6 16 cb 8f bb 25 cf d9 43 8f fa b1 72 54 67 75 b5 e8 2d 04 
cb a9 44 06 8a 21 cc f5 9c fd 39 6e 8c e4 2e 11 83 2a 6b 77 ae 4a 2f 2f 4f 39 4c ca d8 b2 ea 45 
6e d4 e0 d4 4e 8c e7 4d 40 dd f3 23 0e d1 93 b4 9d 6b 7a 5b 8a af 1b 8e bc 98 b3 0d 04 c1 79 38 
b7 48 32 d9 ef e9 f8 32 67 24 71 f7 ce 0f 19 2e 6f 5e 2a ee 2c a7 33 f6 c9 25 1d ad 4f 9e 5d 45 
71 07 1e 6a 44 ce 90 69 ca f3 a0 1b 63 7e 8d 13 72 cf f5 73 ec e3 9c d1 d2 ba f1 4f b8 b1 9f 87 
0f 00 41 00 fc 35 8d 22 f1 c0 9a 6b c0 1f dd 51 80 80 f3 76 a0 01 e9 dd 54 de 50 2e 33 47 99 ce 
6d 16 10 0e cd 39 bb e4 b1 37 88 4b 9c 70 82 dd 04 7d 8c e5 ca 8e 2a f5 f2 0f c7 bc 8c bd 35 38 
e3 a5 37 86 73 05 f2 13 b9 cf 8f 07 d7 f8 fc d8 cc 9f 49 ec 18 c5 16 f7 b3 74 96 67 89 89 58 d6 
62 9b fc 84 40 62 f4 aa 08 f7 34 d3 17 3d ab 76 86 85 0b a9 86 19 55 4c e0 c9 3d 47 0f a6 4f 65 
5a 0c be b1 5a 42 8e e5 c9 e9 68 32 11 d0 3a bb 42 6e 8e 27 17 08 6d ef d0 06 ed 57 c4 01 4e ef 
13 7a d6 a1 bd b9 f1 ac eb b9 0f d8 d8 56 77 eb a0 ef bc 34 51 e1 83 97 7e 9c 6b 55 7c af 65 c3 
d4 83 58 8d af 73 4f 9c 6e b6 2f 80 f2 5c 52 ab 5f 25 4f 68 5c 93 50 f8 84 e7 12 8e 4e 5e 36 dc 
fb 5b f5 56 6c 80 76 fd c5 99 b9 28 b7 ac 16 f1 03 28 4d 5e 60 47 ec 2d dd d1 6a f1 32 5b ce d6 
0f 00 42 00 fc 8d b0 65 3c 61 ff e3 c6 6a ee 8c d8 55 ed da d0 0b 63 1a 32 c4 9f 30 2e e3 ff 43 
0d a4 cd 5a 6e 76 9f b2 51 06 74 3c c0 61 44 73 c5 5e f3 3f 6c ad 18 0e 3d 14 e8 ca 8d ea e7 ad 
dc 9a 8d 77 47 6d 66 ff 37 3e 1a 60 0c 1f 8e c5 e7 fb 50 bd 59 7f e9 cc 8e 6e 13 00 a0 b0 a7 f3 
2b 15 ec dc 11 73 4e ac 5c 78 87 22 60 15 2c a5 6a 13 69 44 f8 62 11 42 66 33 97 66 69 6a 51 a1 
a9 20 42 44 31 f1 aa 7f 4c fb 75 68 3a 62 62 6a 42 05 47 1d 11 cf 22 1b b3 d3 94 ee 85 a8 db e1 
37 c9 88 45 55 00 e6 76 a5 1b 99 b8 c3 67 95 03 a5 f2 e9 3a cb e6 05 4d 19 a0 db d6 65 92 69 c8 
14 04 eb 65 89 28 36 18 c0 a0 be 34 96 c2 e1 85 c1 b0 d3 dd a2 11 73 b5 81 c2 c9 e6 d0 da 10 f6 
06 15 19 c1 0b 7a 0b 23 b3 6b d9 ba ad 97 a1 3d f3 9b a1 ea 91 a7 d0 cb 8b d3 22 be c1 f2 18 d9 
0f 00 43 00 fc fb 78 8c 21 fc f7 b2 63 1c 04 43 91 b0 37 93 11 65 74 77 0a e9 4e d8 da fa cc e5 
32 73 7e e8 14 48 44 7a e1 05 fd 1f 72 bd 9d 0c ad 17 02 c7 b1 fa c7 49 b2 6e 70 08 5d 58 43 b3 
[... cut ...]
41 13 4d e0 6e 87 54 ad a5 41 fb d9 8d c8 e3 69 05 7d dc 31 bd 2e 6e a1 80 cf b7 71 b7 1e 6b 1c 
0f 00 44 00 fc df 90 2a aa 1b 89 d6 36 fb 78 7e 75 dd 23 a6 32 9e 02 f5 94 c8 f2 8a e5 79 43 41 
90 a5 91 cd 1f 13 d0 5f e1 b8 c7 2a 14 99 78 a3 d8 51 40 f4 c2 54 5d c7 92 ab 56 dc be c7 03 f3 
b9 b3 cb 6a 27 a5 8d 63 35 cf a1 c1 b8 e8 4d 6c 7b 90 1e 53 19 56 77 ea b3 ba 2d ab 03 99 86 9a 
96 e0 90 98 91 98 87 ce 74 b7 1b 68 72 7f 2b a9 c8 8f 9b 39 ca 98 6b 79 ␀�␂�*�d3 d5 82 4f 39 7a 
6d 29 cc 34 
0f 00 0f 00 0f 00 0f 00 0f 00 0f 00 0f 00 0f 00 0f 00 0f 00 0f 00 0f 00 0f 00 0f 00