SmingHub/Sming #693: DNSCaptivePortal example does not work for me

icodk, 2016-04-11:
Compiled fine but does not work as expected.
Get response if explicitly write the IP address in the browser but this is missing the idea of Captive portal.
Looks like the onDefault function  is not wired to anything.
What am I missing ?

patrickjahns, 2016-04-11:
It works on my side on a windows laptop aswell as android smartphone without any problem. No matter what domain I enter, I always see the correct page.

In order to be able to help with your issue, please describe your environment as close as possible and ideally a log with debug enabled.

icodk, 2016-04-11:
Yes it works also for me on IPad.  Initially tested only on Windows 10. Still I can't get it to work on windows 10. I  disconnect my Windows PC from wired LAN then connected wifi to "SMING captive portal"
SSID .Then I opened a browser (Chrome), enter dr.dk in the address line, after a few moments Chrome says: There is no Internet connection and that's it.
BTW how does the onDefault function is being called ? I can see it is called when I test with IPad.
Thanks
Will be back tomorrow (can't sleep at work:-)

patrickjahns, 2016-04-12:
You are correct - `server.setDefaultHandler(onIndex);` points to `onIndex` instead of `onDefault`. But it shouldn`t matter when you try the basedomain without paths for lookup (as you ipad correctly handles it).

Some remarks:
Windows since some versions tests the internet connection by trying to connect to a certain domain and looking 'ncis.txt' - if it doesn`t find it - it assumes that there is no internet connection. 
(Similar function in android with /generate_204) 
This example is by no means meant to be a use it and everything you ever wish to achieve with a dns captive portal will be set for. It is a means to demonstrate how the implemented DNS server can be put to use

Further remarks:
Windows keeps a DNS cache - and the cache entries only expire/get removed when TTL is reached. Any webpage that is previously visited and the domain is still in cache will simply fail, since it won`t be able to connect to the previously resolved ip address. 
Either thr entering a non visited domain (even test.test works) - or clear your dns cache ( ipconfig /flushdns in a windows command shell)

icodk, 2016-04-12:
Thanks for the head up. Will try to follow your advice.

hreintke, 2016-04-13:
@patrickjahns : 
Is there a way to set the TTL in the captive portal so that applications have the possibility to force the remote DNS to refresh more frequently.

patrickjahns, 2016-04-13:
The dnsserver allows for changing the ttl. But only for the domains the
server resolved. I don't know if it is possible to clear the cache of
clients before any request is made.

hreintke, 2016-04-13:
@patrickjahns : 
Changing the ttl ot the resolved domains is what I mend. 
We used lowing ttl for a period around updating large web applications and make sure that clients request the new page  instead of the cached old.

It's up to the client (but also routers/proxys etc) to clear the cache regularly.

patrickjahns, 2016-04-29:
@icodk 
any news/results?

@hreintke 
I can lower it - what value do you recommend? 

I will update the example then and maybe add some more notes in the source or a readme

icodk, 2016-04-29:
@patrickjahns 
Had other features to follow. Will return to in future iteration

hreintke, 2016-05-23:
@patrickjahns : 
I think I mixed up diferent ttl's.
Closing the issue for now.

closed 2016-05-23 by hreintke