SmingHub/Sming #695: Add a "powered by Sming" section in the readme.md for project showcase

robotiko, 2016-04-13:
The idea is to link to "Sming powered" projects.

alonewolfx2, 2016-04-13:
agreed. i will ad it soon. thanks for idea

alon24, 2016-04-13:
can u suggest default text?

patrickjahns, 2016-04-13:
@alon24 
what do you mean by default text?

I talked with robotiko and my idea was something along the lines of just having a small section linking to projects that utilized the Sming project.

I think it`s a good addition to the samples to see some real working code with Sming. And also to increase awareness, that Sming can actually be used in "production projects"

alonewolfx2, 2016-04-13:
@robotiko @alon24 @patrickjahns i am already using 3 different project with sming core. i will share and link it but for now its just in Turkish language. i will add in English and share it

patrickjahns, 2016-04-13:
Shall we keep this issue as a space for collecting some links and then deciding which are helpful to provide via the README?

alon24, 2016-04-13:
I thought u were talking about adding text to my Readme for sming,  I am
using sming in all my projects and have been adding a sming Ling to each

robotiko, 2016-04-13:
@alon24 @alonewolfx2  @patrickjahns 

To keep it simple, the idea is to add  a section in Sming readme.md where we add external projects. 
Each entry must have project name and link , followed by a small description  (brief:1-2 lines, English please ;) )  about what it does.
 
As example this repo just keeps a list of interesting projects and has been around for some time.  
https://github.com/sindresorhus/awesome-nodejs

and example of the PR to add a new one: 
https://github.com/sindresorhus/awesome-nodejs/pull/487
It just involves changes in the main Sming readme.

patrickjahns, 2016-04-22:
@robotiko 

I like this approach - maybe we can collect some for the initial PR in this Issue? 

Another option would be to use the Wiki for a table with projects? But it has the downside, that not everyone can contribute right?

hreintke, 2016-05-23:
Issue should be moved to SmingRTOS.

closed 2016-05-23 by hreintke