SmingHub/Sming #699: Conflict between RtcClass and RBoot RTC data

capablegh, 2016-04-15:
File 'Sming/SmingCore/Platform/RTC.h' defines RTC_DES_ADDR as 64, the address for 'RtcData'. And file 'Sming/rboot/rboot.h' defines RBOOT_RTC_ADDR also as 64 for 'rboot_rtc_data' when BOOT_RTC_ENABLED is defined. Noted in Git commit '49f786e35f5c35305b97d43165cdd97ec8c36cc5'

Opening request per [http://www.esp8266.com/viewtopic.php?f=35&t=9532#p45515](url)

raburton, 2016-04-15:
You can change one of these values to use both at once (the rBoot one is not enabled by default), but there does need to be a better system here. Possibly just allowing one or the other to be overridden with a define in the CFLAGS to enable to user to decide where to put it. It'd be easier to make the overridable one the RTC class because rBoot one once deployed in the bootloader needs to stay the same in the API.

hreintke, 2016-05-23:
Will not be updated in SmingNONOS.
If issue still exists, open an issue in SmingRTOS

closed 2016-05-23 by hreintke