SmingHub/Sming #700: Any possibility of reducing RTC memory in rboot?

capablegh, 2016-04-15:
The available RTC memory is very small, and it would help to leave as much possible for application to use. Would the following questions provide the needed considerations?
- When there is a checksum is the magic really needed?
- When possible can fields be made bit fields instead of uint8 or larger? For example: the next and last modes, as well as the last and temp rom, all together be made to use only 8 bits together.
- Can "#pragma pack(0)" be used to ensure memory use is compact.
Doing the above could reduce the RTC mem usage by rboot to just two bytes instead of 9.

raburton, 2016-04-15:
> When there is a checksum is the magic really needed?

Checksum is only 8 bits, so in random data you'd have a 1 in 256 chance of the checksum being correct, but add the magic so we know we are at least working with an rBoot rtc structure and you're a bit better off. A 32bit checksum would be better, and you could do away with the magic, but that wouldn't save you a lot of space and wouldn't be as logical as first checking we have an rBoot structure with the magic, then validating the contents with the checksum.

> When possible can fields be made bit fields instead of uint8 or larger? For example: the next and last modes, as well as the last and temp rom, all together be made to use only 8 bits together.

No they can't - the rom slot has a range of 0-255, so both of these need to be 8 bit. The flag fields carry more than two options, so cannot be reduced to a single bit each. True, they don't currently need 8 bits each, allowing some room for expansion you could merge the two together and have 4 bits each, but again that's only saving you a byte.

> Can "#pragma pack(0)" be used to ensure memory use is compact.

Would this make any difference? Compiles for me without any wasted space, using gcc.

> Doing the above could reduce the RTC mem usage by rboot to just two bytes instead of 9.

I only see a realistic saving of 2 bytes, so at present I don't see a good reason to make any changes, especially  as this would break compatibility with any existing users - rBoot would need to be updated as well as the api and I don't recommend anyone update the bootloader via ota, so devices would need to be recalled to serial flash them.

capablegh, 2016-04-15:
Thanks for your response..

FWIW, the pragma compiler directive only tries to ensure that all non byte access are aligned as specified. It could have made a difference if bit fields were used. It's a non issue if the byte/half words/words are aligned manually, as in this case.

There's still enough space in the user part of the RTC memory, so not enough justification pursue any change.

rBoot is an excellent value add to Sming, thanks for making it available!

closed 2016-04-24 by hreintke