SmingHub/Sming #701: Possibly improving RTC memory management

capablegh, 2016-04-15:
At this writing the rboot and RTC subsystems use the RTC memory. Applications may need some memory also. So, instead of each subsystem using it's own checksum and a magic number, why not create a new class that manages the RTC memory. All subsystems needing to use the RTC memory write to it using this RTC memory manager class methods.

Not only could the RTC memory utilization be made more compact, it will also give a single place to name the IDs for each RTC memory using subsystem. It could also simplify the logic in these subsystems.

raburton, 2016-04-15:
So you basically want to make a file system for the rtc data? That's going to use more space, not less. Also, rBoot is a third party component integrated unmodified into Sming, changes to it should be requested upstream. However, a change like this - an application framework specifc integration, incompatible with all other rBoot users, and breaking existing deployments - will not be accepted upstream.

closed 2016-04-15 by raburton

capablegh, 2016-04-15:
> So you basically want to make a file system for the rtc data?
Not really a file system, but a common means to access the RTC memory. This may be user application too. Essentially, this would allow for a common checksum32 instead of each one having it's own magic number and/or checksum.

Beside, the request is really to the Sming framework not just rBoot. Any how, like I said it's a suggestion/consideration. Thanks for responding.