SmingHub/Sming #707: How to get a 64k/128k SPIFFS?

susie-derkin, 2016-04-18:
Hi,

because of the fact that I only have 512k ESPs right here, I try to limit the SPIFFS to 128k or better 64k.
I found in the file Makefile-projekt.mk the lines for configuring the SPIFF_SIZE and tried to use the settings

	flashimageoptions += -fs 2m
	SPIFF_SIZE ?= 131072  #128K

which shall be ok for 256k ESPs.
But with those settings my ESP will not start up - spiffy is telling me that the filesystem is really of the configured size and starts at the common location (same as the 192k option):
`# Generating spiffs_bin
Checking for spiffs files
files directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x20000 (131072) bytes.
Adding files in directory 'files'.
Added 'index.html' to spiffs (600 bytes).
Added 'style.css' to spiffs (255 bytes).
out/firmware/spiff_rom.bin---------->0x4C000`

Is there a way to get a 128k or better a 64k SPIFFS? What settings are the correct ones for the makefiles?
Thanks in advance

hreintke, 2016-04-24:
@susie-derkin : 
No need to update makefile-project. 
You can add the  SPIFFS_SIZE line in the makefile-user from your application, or as a environment variable to use one setting for all your applications.
The spiffs location is only dependent on the application size, not on the spiffs size.

The "out/firmware/spiff_rom.bin---------->0x4C000' shows the startaddress and will change according to application size,

susie-derkin, 2016-04-24:
@hreintke:

Thanks for your answer, I just tried the suggestion by putting SPIFF_SIZE ?= 65536 in Makefile-user.mk.
Make delivers the following output, everything seems ok:
`# Generating spiffs_bin
Checking for spiffs files
files directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x10000 (65536) bytes.
Adding files in directory 'files'.
Added 'index.html' to spiffs (600 bytes).
Added 'style.css' to spiffs (255 bytes).
out/firmware/spiff_rom.bin---------->0x4C000`

But in my init code I have the following code to check the SPIFFS size:
	system_print_meminfo ();
	uint32_t size = flashmem_get_size_bytes();
	spiffs_mount(); // Mount file system, in order to work with files
	Serial.println("Flash memory size = " + String(size));

	spiffs_config cfg = spiffs_get_storage_config();
	Serial.println("SPIFFS size = " + String(cfg.phys_size));

And here it tells me, that SPIFFS has still a size of 192kB:

	data  : 0x3ffe8000 ~ 0x3ffe8374, len: 884<CR><LF>
	rodata: 0x3ffe8380 ~ 0x3ffeaf9c, len: 11292<CR><LF>
	bss   : 0x3ffeafa0 ~ 0x3fff2240, len: 29344<CR><LF>
	heap  : 0x3fff2240 ~ 0x3fffc000, len: 40384<CR><LF>
	fs.start: size:192 Kb, offset:0x4c000<LF><CR>
	<LF>
	mount res: 0<LF><CR>
	<LF>
	Flash memory size = 524288<CR><LF>
	SPIFFS size = 196608<CR><LF>

What is wrong with this code? Do you have any hint?
Thanks in advance

hreintke, 2016-04-25:
@susie-derkin : 
You hit a limitation of the spiffs implementation of SmingNONOS. The application is not able to detect the size of the FS from the information on the flash. That is why it makes an educated guess, depending in the SPI_SIZE. A way to get around this limitation is using the spiffs_mount_manual in which the application defines the starting address & spiffs_size. 
The starting address is the 'out/firmware/spiff_rom.bin---------->0x4C000' from the output. The size is your defined SPIFFS_SIZE. 
Beware that the starting address is not fixed but dependent on the application size.

Easier option is to move your application to SmingRTOS. The spiffs used in there and application can mount filesystems which size is taken from flash info not guessed.

susie-derkin, 2016-04-28:
@hreintke 
Again thanks for your answer, you gave me the necessary hint and I tried it with spiffs_mount_manual. But here I can tell, that the starting address parameter is not the value you get from spiffy (I tried it it and it will fail), the starting address has to be  INTERNAL_FLASH_START_ADDRESS + (the value where spiff_rom.bin starts). The INTERNAL_FLASH_START_ADDRESS value is defined in flashmem.h.

If the INTERNAL_FLASH_START_ADDRESS is not added, you get something like:
fs.start: size:64 Kb, offset:0xbfe4c000         <--- this is the wrong offset resulting in an flash_read error
ERROR: ERROR in flash_read: r=1 at 0004c000 

Thanks again, I now really have a 64k SPIFFS!

hreintke, 2016-04-29:
@susie-derkin : 
Within Sming you indeed need to add the flash start address, I knew that and forgot to mention it.
Sorry for that. 
When you move to SmingRTOS, you need to "remove" the adding. We decided to use real flash addresses there.

closed 2016-04-29 by hreintke