SmingHub/Sming #708: Latest chocolately install has wrong SDK version

ehughes, 2016-04-20:
Hello:

I followed the instructions here:

https://github.com/SmingHub/Sming/wiki/Windows-Quickstart

I am getting build errors such as:

c:\tools\sming\sming\smingcore\platform\station.h:36:25: error: 'SC_TYPE_ESPTOUCH_AIRKISS' was not declared in this scope
  SCT_EspTouch_AirKiss = SC_TYPE_ESPTOUCH_AIRKISS
                         

From what I have read this is due to a wrong version of the ESP SDK.

Can you please take a look?

patrickjahns, 2016-04-24:
@hreintke  @alonewolfx2 

this is related to the question I had in gitter - is it possible to host the chocolate package in a SmingHub repository? Then it would be easier to keep track of changes and accordingly update it without relying on one single person?

hreintke, 2016-04-24:
@patrickjahns : 
I totally agree that relying on a single person needs to be avoided. 
For the chocolat package : Is that "just a textfile with links and dependencies" or also binaries

alonewolfx2, 2016-04-24:
@kireevco @hreintke i maked a PR to choco repo for updating udk to 2.0.9 again for sdk 1.5 compatibility. 
also i am thining to move choco pacages repo to sming repo. any idea?

hreintke, 2016-04-24:
@alonewolfx2 : 
I don't know whether the choco repo is "sming only". If not, we should not move to SmingHub.
When number of choco files is low, it might be better to add them to the Sming/SmingRTOS in a specific directory.

hreintke, 2016-06-21:
@alonewolfx2 : 
Do you have the knowledge/acces rights to update the choco installation and provide additional documentation for that ?

slaff, 2017-07-27:
The workaround in order to get the latest stable version of Sming is

```
choco upgrade -y -f sming.core --version 0.0.4
```

If someone knows a place to host free of charge choco packages then we can update also the other Sming packages.

slaff, 2017-07-29:
That should be fixed using the new source. See the modified WIKI page: https://github.com/SmingHub/Sming/wiki/Windows-Quickstart for details.

closed 2017-07-29 by slaff