SmingHub/Sming #711: Persisting data stored in spiff through OTA

icodk, 2016-04-23:
OTA enables you update also the file system which is a great feature. However if you wants to preserve data, stored in the existing file system, there is no obvious way how to do it. In case of settings file like in the .settings.conf from the HttpServer_ConfigNetwork example, you could post all values  to a data base and then read it back after OTA. this will require suitable infrastructure. If anybody has a better idea please share it with us.

patrickjahns, 2016-04-24:
Hey,

here are some thoughts on it:
Question is, for what reason do you need to update the spiffs? Is it just to update some (web)files? 
Then it would also be possible to write a seperate class, that downloads these files and replaces the already existing ones.

I am using that approach in a project of mine:
https://github.com/patrickjahns/esp_rgbww_firmware/blob/master/app/otaupdate.cpp

Next thoughts:
- if it is just configuration data, and the data is small, there is space in rtc 

Also an idea:
Keep that information in memory, wait until the spiff is updated, mount the new spiffs, write the configuration

hreintke, 2016-04-24:
If the data is small I would suggest the "memory" solution.

For larger data blocks : 
- The Rboot area has a size of one flash sector (4096 bytes) On a small part of it is used by the rboot functionality.  The rboot functions set_config and get_config are taking care of preserving rboot/userdata when updating. -> Using that ii is outside the spiffs.

- In SmingRTOS there will be a flashstream option shortly. Using that you can read/write any amount of data to any location of the flash. It is a stream and not a filesystem, some care needs to be taken when using it.

icodk, 2016-04-24:
@patrickjahns, @hreintke 
Thanks for the input. The memory idea is perfect for me. I only have to call save() just before rebooting in
OtaUpdate_CallBack()

closed 2016-04-24 by hreintke