SmingHub/Sming #715: rboot - makefile missing options

patrickjahns, 2016-04-27:
I have tried getting the temprom feature to work, but rboot never "enables" the rtc module

Serial output from booting
```
16:45:23.591> load 0x3ffe8000, len 660, room 12 
16:45:23.591> tail 8
16:45:23.591> chksum 0xcd
16:45:23.591> csum 0xcd
16:45:23.591> 
16:45:23.591> rBoot v1.3.0 - richardaburton@gmail.com
16:45:23.591> Flash Size:   32 Mbit
16:45:23.591> Flash Mode:   QIO
16:45:23.591> Flash Speed:  40 MHz
16:45:23.591> rBoot Option: Big flash
16:45:23.591> 
16:45:23.803> Booting rom 0.
```

Makefile-user.mk
```
# use rboot build mode
RBOOT_ENABLED = 1

# 
RBOOT_RTC_ENABLED = 1

# enable big flash support (for multiple roms, each in separate 1mb block of flash)
RBOOT_BIG_FLASH = 1

# two rom mode (where two roms sit in the same 1mb block of flash)
RBOOT_TWO_ROMS  = 0

# where does the filesystem reside
RBOOT_SPIFFS_0  = 0x100000
RBOOT_SPIFFS_1  = 0x300000 
```

Relevant section from makefile
```
make all V=1 
mkdir -p out/build/app
mkdir -p out/build//C/opt/Sming/Sming/rboot/appcode
mkdir -p out/firmware
OC out/build/libmain2.a
/usr/bin/make -C /C/opt/Sming/Sming/rboot
make[1]: Entering directory `/C/opt/Sming/Sming/rboot'
CC rboot-stage2a.c
xtensa-lx106-elf-gcc -Os -O3 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals  -D__ets__ -DICACHE_FLASH -DBOOT_BIG_FLASH  -c rboot-stage2a.c -o /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/build/rboot-stage2a.o
LD /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/build/rboot-stage2a.elf
xtensa-lx106-elf-gcc -Trboot-stage2a.ld -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/build/rboot-stage2a.o -Wl,--end-group -o /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/build/rboot-stage2a.elf
E2 /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/build/rboot-hex2a.h
C:\opt\esptool2\esptool2.exe -quiet -header /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/build/rboot-stage2a.elf /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/build/rboot-hex2a.h .text
CC rboot.c
xtensa-lx106-elf-gcc -Os -O3 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals  -D__ets__ -DICACHE_FLASH -DBOOT_BIG_FLASH -I/c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/build  -c rboot.c -o /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/build/rboot.o
LD /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/build/rboot.elf
xtensa-lx106-elf-gcc -Teagle.app.v6.ld -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/build/rboot.o -Wl,--end-group -o /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/build/rboot.elf
E2 /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/firmware/rboot.bin
C:\opt\esptool2\esptool2.exe -quiet -bin -boot0 -4096 -qio /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/build/rboot.elf /c/Users/patri/@projects/arduinoprojects/RGBCTRLfw/out/firmware/rboot.bin .text .rodata
make[1]: Leaving directory `/C/opt/Sming/Sming/rboot'
```

Comparison to make for application
```
/C/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -Iapp -I/C/opt/Sming/Sming/rboot/appcode -Iapp/include -I/C/opt/Sming/Sming/rboot/appcode/include -Iinclude -I/C/opt/Sming/Sming/include -I/C/opt/Sming/Sming/ -I/C/opt/Sming/Sming/system/include -I/C/opt/Sming/Sming/Wiring -I/C/opt/Sming/Sming/Libraries -I/C/opt/Sming/Sming/SmingCore -IC:/Espressif/ESP8266_SDK/../include -I/C/opt/Sming/Sming/rboot -I/C/opt/Sming/Sming/rboot/appcode -IC:/Espressif/ESP8266_SDK/include -Wpointer-arith -Wundef -Werror -Wl,-EL -nostdlib -mlongcalls -mtext-section-literals -finline-functions -fdata-sections -ffunction-sections -D__ets__ -DICACHE_FLASH -DARDUINO=106 -DGITVERSION=\"v0.2.5-5-gdec4-dirty\" -DGITDATE=\"2016-04-25\" -DFWVERSION=\"`cat VERSION`\"  -Os -g -DBOOT_BIG_FLASH -DSPIFF_SIZE=224288  -DRBOOT_SPIFFS_0=0x100000 -DRBOOT_SPIFFS_1=0x300000  -DRBOOT_INTEGRATION -DBOOT_RTC_ENABLED -fno-rtti -fno-exceptions -std=c++11 -felide-constructors -c app/application.cpp -o out/build/app/application.o
```

Looking at the make log - it seems that the Makefile inside rboot folder does not aknowledge the options specified. 
Is this targeted behavior? Any reason why it is not included or just a "bug"

patrickjahns, 2016-04-27:
Also noting, rboot.c has several more options which are not accessible from withing Sming:

```
#ifdef BOOT_BIG_FLASH
	ets_printf("rBoot Option: Big flash\r\n");
#endif
#ifdef BOOT_CONFIG_CHKSUM
	ets_printf("rBoot Option: Config chksum\r\n");
#endif
#ifdef BOOT_GPIO_ENABLED
	ets_printf("rBoot Option: GPIO mode\r\n");
#endif
#ifdef BOOT_RTC_ENABLED
	ets_printf("rBoot Option: RTC data\r\n");
#endif
#ifdef BOOT_IROM_CHKSUM
	ets_printf("rBoot Option: irom chksum\r\n");
#endif
```

raburton, 2016-04-28:
Ideally these options would be set in rboot.h, but because of the way sming uses them some of the options have been enabled for passing through via make. Not all options have been made available int his way in rBoot, although there is not reason they can't be, please submit a suitable patch upstream.

closed 2016-11-15 by slaff