SmingHub/Sming #717: LWM2M client library

cvgaviao, 2016-04-27:
It would be great to manage devices based on ESP8266 using LWM2M manage applications as the one provided by the Leshan Eclipse Project [1].

The Wakaama Eclipse project [2] provides a C based implementation files (built with cmake) that is aimed to facilitate the port for any device. But I don't know if it would be feasible to integrate it with Sming. Perhaps any commiter would evaluate and give some feedback ?

[1] - https://github.com/eclipse/leshan
[2] - https://github.com/eclipse/wakaama

hreintke, 2016-04-29:
@cvgaviao :
Supporting multiple managing frameworks would be good for Sming.
There are some (smaller) implementations on websocket and mqtt. 
You can take a look at those to identify whether if is feasible to port wakaama.

hreintke, 2016-05-23:
No activity, closing issue.
If further information is available, submit an issue/PR in SmingRTOS.

closed 2016-05-23 by hreintke