SmingHub/Sming #718: (temp) rom not booting after OTA

patrickjahns, 2016-04-28:
Already mentioned it in gitter - but started to investigate the issue a bit more and provide further logfiles.

I have a strange issue with OTA via `rBootHttpUpdate`. Everything works fine - when I use the same rom when flashing and updating via OTA. 
If I flash one rom - and then modify the source slightly (i.e. remove a char from a debug message), rebuild the source and use the new rom for OTA the new rom would not boot.

This is strange behavior and I am not sure how to trace this issue further. Any help is highly appreciated.

More Information:
Makefile-user.mk rboot settings:
```
#### rBoot options ####
# use rboot build mode
RBOOT_ENABLED = 1

#  enable tmp rom switching
RBOOT_RTC_ENABLED = 1

# enable big flash support (for multiple roms, each in separate 1mb block of flash)
RBOOT_BIG_FLASH = 1

# two rom mode (where two roms sit in the same 1mb block of flash)
RBOOT_TWO_ROMS  = 0

# where does the filesystem reside
RBOOT_SPIFFS_0  = 0x100000
RBOOT_SPIFFS_1  = 0x300000 
```

Logfile (had to catch the parts with several attempts since Baudrate in application is 115200 - with rBoot/espressif it is 74880
 
rBootHttpUpdate success message
```
Firmware download finished!
 - item: 0, addr: 2000, len: 339296 bytes
 - item: 1, addr: 100000, len: 786432 bytes
Firmware updated.
ApplicationOTA::rBootCallback
ApplicationOTA::rBootCallback temp boot 0
```

System.restart():
```
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 1948, room 16 
tail 12
chksum 0x37
ho 0 tail 12 room 4
load 0x3ffe8000, len 772, room 12 
tail 8
chksum 0x6c
csum 0x6c

rBoot v1.3.0 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting temp rom.
Booting rom 0.
```

Some unreadable chars - and then with Baudrate 74880 I could catch the following

```
E:M 36
E:M 16
E:M 16
E:M 16
E:M 16
E:M 16
E:M 16
E:M 32
E:M 756
E:M 21
E:M 21
E:M 21
E:M 21
E:M 21
E:M 21
E:M 21
E:M 21
E:M 21
E:M 21
E:M 21
E:M 21
E:M 21
E:M 21

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 1948, room 16 
tail 12
chksum 0x37
ho 0 tail 12 room 4
load 0x3ffe8000, len 772, room 12 
tail 8
chksum 0x6c
csum 0x6c

rBoot v1.3.0 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 1.
```

Rom 1 starts without problem - I tried to find any explanation for `E:M XXX` message but haven`t found any. 

Linker file for rom is here:
https://gist.github.com/patrickjahns/4ae4564a31cd744b39d18a5dbf6116ce

raburton, 2016-04-28:
I think I see what's going on here, looks like a bug when using bigflash and rtc temp boot feature at the same time. You're getting the second rom memory mapped even though you've booted the first one. The memory mapping code reads the config sector to see which rom is selected and so which rom to mamory map, but that doesn't take account of the rtc temp mode. Checking for this will needed to be included in rboot-bigflash.c - checking rtc data (if enabled) +/- fallback to reading config or just reading config if rtc not enabled. I only have internet access at work at the moment (in a hospital) so can't really do any dev here. Are you able to propose a fix and submit it as a PR upstream? Otherwise it'll have to wait till I get internet access back which might be a week or two.

patrickjahns, 2016-04-28:
@raburton 
Thanks for having a look at it. 
Just for clarification - this does not happen when using the exact same roms for flashing via serial and OTA afterward - it will switch/boot properly to temp rom fine.
Problem only occurs when OTA rom differs from the initial rom flashed via serial

I will have a look and see if I can create a fix.
Take your time and if I can`t fix it - I will gladly wait - no

raburton, 2016-04-28:
Yes, same rom would work fine, because one rom's iram code will be accessing the other roms irom code which has been memory mapped by mistake. If the roms are the same then that will work fine, because they are mapped at the same memory address and functions within it will be located at the same position. A recompiled one would still be mapped to the same address, but the function offsets within the rom will be different.

patrickjahns, 2016-04-28:
Thanks for the feedback - been experimenting and I came up with this:
https://gist.github.com/patrickjahns/183f72f7dc4d6c5fcc65b6150141a18e

My remarks:
- it would probably be better to not declare system_rtc_mem and calc_chksum in rboot-bigflash.c again
- `rtc.next_mode` is already changed and I needed to compare `rtc.last_mode` to check if it is tmp rom booting.

As per your suggestion this works, but in general this feels a bit hackish and maybe you have a better solution/idea for it?

raburton, 2016-05-03:
@patrickjahns 

Agree that best not to duplicate these function, but really we need to create smaller more optimized versions of them specific for use here. This function has to reside in iram and so we need to keep it as small as possible. Also, no need to test of it's a temp boot, for normal boot (with rtc enabled) it will record the booted rom in the same place, so we can just use that.

How about this: https://gist.github.com/raburton/88312dc1f0d432f7b31c5cf9b20ac6c2
It's dirty, but it's only a few bytes longer than the non-rtc method. Doing it properly (checking the checksums, etc) makes the code much larger and shouldn't be necessary, I know that's asking for trouble, but we just can't afford to use the ram. Only compile tested, do to no device access at the moment. Have a look, hopefully it works and the code makes sense enough for you to see the compromises (and to fix it if it doesn't work first time!).

raburton, 2016-05-06:
@patrickjahns just wondered if you'd had chance to look at the code above...

patrickjahns, 2016-05-06:
@raburton 

I am out of town till weekend - will test it then.

patrickjahns, 2016-05-08:
@raburton 

the implementation in your gist fails to boot with repeating Exception message:
```
15:28:39.687> Fatal exception (0): 
15:28:39.687> epc1=0x40202a18, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
15:28:39.687> Fatal exception (0): 
15:28:39.687> epc1=0x40202a18, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```

When I reset the esp , I receive random chars - haven`t found a baud rate that works in order to find something useful from the characters

When testing the version of mine, everything works as expected.

How is the rom selected on the first ever boot after a flash? Might it be that the RTC data is missing or not correct?

patrickjahns, 2016-05-08:
I`ve tested around a bit by stripping out some parts of the previous example and aligned it with your version. Except the data retrival is done via system_rtc_mem 

This works without issues:
https://gist.github.com/patrickjahns/183f72f7dc4d6c5fcc65b6150141a18e

To me it seems that something with the minified reading of data is off (address/calculation?)

raburton, 2016-05-08:
Thanks, yes there is probably a bug in there because I haven't been able to
test it yet. I'll have internet access back and a couple of hours free
tomorrow evening to have a look at a couple of outstanding issues.

Richard.

patrickjahns, 2016-05-09:
I`ve been testing around a bit - and here is another strange observation I made. If I further strip out the checking parts in the code, I will also end up with exceptions.

```
uint32 system_rtc_mem(int32 addr, void *buff, int32 length, uint32 mode) {

 int32 blocks;
 // copy the data
    for (blocks = (length >> 2) - 1; blocks >= 0; blocks--) {
        volatile uint32 *ram = ((uint32*)buff) + blocks;
        volatile uint32 *rtc = ((uint32*)0x60001100) + addr + blocks;
		if (mode == RBOOT_RTC_WRITE) {
			*rtc = *ram;
		} else {
			*ram = *rtc;
		}
    }

    return 1;
}
```

This seems quite strange to me. If you find the time it would be great to talk about what`s going on there - really would like to understand this behavior a bit more

raburton, 2016-05-10:
Found a couple of little bugs and updated the gist, code now working fine for me (once I remembered to set the flash size correctly when writing, that wasted a bit of time!)

patrickjahns, 2016-05-10:
@raburton 

Tested and can confirm it is working. I`ve tested the same rom as well as different roms updated via OTA 

If you find the time I`d be curious to know how you arrived with the address calculation for the bit. I kind of understand it - but am missing some parts.

raburton, 2016-05-10:
> Tested and can confirm it is working. I`ve tested the same rom as well as different roms updated via OTA

Great, I'll get it merged this evening...

> If you find the time I`d be curious to know how you arrived with the address calculation for the bit. I kind of understand it - but am missing some parts. 

To explain the code:

```
rboot_rtc_data rtc;
uint8 off = (uint8*)&rtc.last_rom - (uint8*)&rtc;
```
So this works out the offset of bytes into the structure for the last_rom byte. Should return 6. The rtc structure is never used, except for this calculation, and so the compiler will optimise this out and just uses the value 6.

```
volatile uint32 *rtcd = (uint32*)(0x60001100 + (RBOOT_RTC_ADDR*4) + (off & ~3));
```
Now we take the address of the rtc data area, add 64*4 to reach the user data area and add the offset from above rounded down to the nearest 4 bytes (as we have to read 4 byte aligned), so that's 4. The compiler actually optimises this all out too and just stores the final resulting address into the code.

```
val = ((uint8*)&val)[off & 3];
```
After we have copied out the 4 bytes, we treat this uint32 as a uint8[4] and use the remainder of offset/4 to access the byte of interest.

This all results in pretty short code, which is what we need as it's going to be in iram and will detract from what the user has available for their own code. The downside is that there is no validation of the checksum and if there is a problem we'll probably crash. This should never happen, but those are the cases when you can get caught out if for some reason it does ever happen. However in this case it's a deliberate decision to take that risk.

closed 2016-05-11 by raburton