SmingHub/Sming #720: out/firmware/0x09000.bin file not generated (Error 2)

DominicMe, 2016-04-29:
I am trying to setup Sming under Arch Linux but I am stuck. I had it working before but I reinstalled and must have missed something in my notes.

Here's how I attempt to compile:

```
cd ~/Sming/samples/Basic_Blink
sudo make flash
```

And here's the error I get:

```
stat: cannot stat 'out/firmware/0x09000.bin': No such file or directory
Killing Terminal to free /dev/ttyUSB0
/home/dom/Sming/Sming/Makefile-project.mk:338: recipe for target 'flash' failed
make: [flash] Killed (ignored)
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0xC000 out/firmware/spiff_rom.bin
usage: esptool write_flash [-h] [--flash_freq {40m,26m,20m,80m}]
                           [--flash_mode {qio,qout,dio,dout}]
                           [--flash_size {4m,2m,8m,16m,32m,16m-c1,32m-c1,32m-c2}]
                           [--verify]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'out/firmware/0x00000.bin'
/home/dom/Sming/Sming/Makefile-project.mk:338: recipe for target 'flash' failed
make: *** [flash] Error 2
```

there are no bin files being created in out/firmware directory.

esp directory: /opt/esp-open-sdk/esptool
esp2 directory: /opt/esp-open-sdk/esptool2
sming directory ~/Sming/Sming/ (also tried installing to /opt/ with same error)

Exports:

```
export ESP_HOME=/opt/esp-open-sdk
export SMING_HOME=/home/user/Sming/Sming
```

What am I missing here?

hreintke, 2016-04-29:
@DominicMe : 
Is the stat program in your path ?

DominicMe, 2016-04-29:
If you mean if I am able to run it then then yes. `stat --help ` gives expected output. I assume stat is installed along with other system packages as I did not have to install it manually.

hreintke, 2016-05-23:
No activity, closing issue

closed 2016-05-23 by hreintke