SmingHub/Sming #722: Build 2 roms for rBoot

samschaerer, 2016-05-02:
Just a question from a dummy...

How can I build 2 roms from one project? I use the _Basic_rBoot_ example from SmingRTOS. Now I would like to have for example an _app0.cpp_ and _app1.cpp_ and then build _rom0.bin_ and _rom1.bin_ from them. 
If possible, I would like also to flash them together to 0x002000 and 0x202000 using make-flash.

Thanks for answer

hreintke, 2016-05-02:
From makefiles that is not implemented at this moment.
Manually it would be the following steps : 
- build app0 -> flash to esp
- build app1 -> fash manually or update the flash line in make-boot.mk for new location

or : 
- build app0
- build app1
flash manually the rom0's from both application directories

samschaerer, 2016-05-02:
Ok, thank you @hreintke

closed 2016-05-02 by samschaerer

raburton, 2016-05-03:
Dual booting two completely different roms is a perfectly valid use case, but not one that is accommodated by the Makefiles (nor is it likely to be added because it would make the Makefiles much more complicated). The simplest way to do it is simply to compile your two separate apps separately, as two separate apps. If they are similar and share a lot of code you could make a custom makefile, use libraries, includes, symlinks, etc.