SmingHub/Sming #723: SPIFF_SIZE does't changes with SPI_SIZE correctly

svireli, 2016-05-03:
When SPI_SIZE is set to 4M SPIFF_SIZE is still 512Kb.
I solved this by changing values in this files.

Makefile-project.mk
[https://github.com/SmingHub/Sming/blob/develop/Sming/Makefile-project.mk#L196](url)
Makefile-rboot.mk
[https://github.com/SmingHub/Sming/blob/develop/Sming/Makefile-rboot.mk#L203](url)

alonewolfx2, 2016-05-03:
what is your os? windows linux?

svireli, 2016-05-03:
Linux.

raburton, 2016-05-03:
> When SPI_SIZE is set to 4M SPIFF_SIZE is still 512Kb.

This is deliberate.

> I solved this by changing values in this files.

If you want something other than the default you can set SPIFF_SIZE manually.

svireli, 2016-05-03:
Thank you. 
I thought SPI_SIZE was required to change size.

raburton, 2016-05-03:
You should still set SPI_SIZE correctly, and you'll get a reasonable default for the SPIFFS_SIZE. But the blocks of code you refer to (which set the default) use `?=` to set SPIFF_SIZE which means it will only be set to the defaults if no value has already been provided, so you can set this variable manually first.

closed 2016-05-03 by raburton