SmingHub/Sming #724: error on apptest.cpp

cvgaviao, 2016-05-03:
I can build Sming properly inside eclipse CDT using make, but there is a file with an error in the line 23. This is the line with error in apptest.cpp:
`	DynamicJsonBuffer jsonBuffer;
`
> The type 'ArduinoJson::Internals::BlockJsonBuffer' must implement the inherited pure virtual method 'ArduinoJson::JsonBuffer::alloc'

hreintke, 2016-05-04:
@cvgaviao :
The apptest.cpp is an obsolete file from early testing of Sming framework.
The application examples are available in the samples direcory.

The apptest.cpp is already removed from SmingRTOS.

cvgaviao, 2016-05-04:
@hreintke  should I remove and propose a PR or you already have done that ?

hreintke, 2016-05-04:
@cvgaviao : 
Yes, please make a PR so this will be solved in SmingNONOS too.

hreintke, 2016-05-23:
Issue resolved

closed 2016-05-23 by hreintke