SmingHub/Sming #727: ArduinoJSON is not working

matoushybl, 2016-05-06:
Hi, I am trying to use ArduinoJSON provided in this project, but whenever I try to use it, my project fails to compile, the output is: 
```
out/build/app_app.a(application.o):(.text._Z17mqttCheckCallbackv$part$7+0x1c): undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
out/build/app_app.a(application.o): In function `mqttCheckCallback()':
/Users/matous/ClionProjects/Pathfinder/app/application.cpp:79: undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
out/build/app_app.a(application.o): In function `_ZN15StringSumHelperC4ERK6String':
/Users/matous/Pathfinder/Sming/Sming/SmingCore/../Wiring/WString.h:273: undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
```

I have also tried the `HttpServer_ConfigNetwork` sample, but the compilation fails too. The output is very similar to the one from my project.

Could you please have a look at it?

Thanks

hreintke, 2016-05-07:
@matoushybl : 
The httpserver_confignetwork example compiles OK with me.
Also when I include a root.getideAt(" .. ") statement.

What version of Sming and SDK are you using ?

matoushybl, 2016-05-07:
Hi, I have pulled the last develop. I am using this SDK: `esp-alt-sdk-v1.5.0.258-macos-x86_64`

hreintke, 2016-05-07:
Did you rebuild the sming library after pulling latest develop ?

matoushybl, 2016-05-07:
Yes, I did.

matoushybl, 2016-05-07:
The sample gives me more errors that the ones with ArduinoJSON, also with `WebSocket` class.

```
LD out/build/app.out
out/build/app_app.a(application.o):(.text._ZN26ApplicationSettingsStorage4loadEv[_ZN26ApplicationSettingsStorage4loadEv]+0x24): undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
out/build/app_app.a(application.o):(.text._ZN26ApplicationSettingsStorage4loadEv[_ZN26ApplicationSettingsStorage4loadEv]+0x28): undefined reference to `ArduinoJson::JsonVariant::asString() const'
out/build/app_app.a(application.o):(.text._ZN26ApplicationSettingsStorage4loadEv[_ZN26ApplicationSettingsStorage4loadEv]+0x9f): undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
out/build/app_app.a(application.o): In function `ApplicationSettingsStorage::load()':
/Users/matous/Pathfinder/Sming/samples/HttpServer_ConfigNetwork/include/AppSettings.h:26: undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
out/build/app_app.a(application.o): In function `_ZN11ArduinoJson9Internals15BlockJsonBufferINS0_16DefaultAllocatorEEC4Ej':
/Users/matous/Pathfinder/Sming/Sming/SmingCore/Network/../../SmingCore/../Services/ArduinoJson/include/ArduinoJson/Internals/BlockJsonBuffer.hpp:36: undefined reference to `ArduinoJson::JsonVariant::asString() const'
out/build/app_app.a(application.o): In function `ApplicationSettingsStorage::load()':
/Users/matous/Pathfinder/Sming/samples/HttpServer_ConfigNetwork/include/AppSettings.h:29: undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
/Users/matous/Pathfinder/Sming/samples/HttpServer_ConfigNetwork/include/AppSettings.h:31: undefined reference to `ArduinoJson::JsonVariant::asString() const'
/Users/matous/Pathfinder/Sming/samples/HttpServer_ConfigNetwork/include/AppSettings.h:31: undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
/Users/matous/Pathfinder/Sming/samples/HttpServer_ConfigNetwork/include/AppSettings.h:33: undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
/Users/matous/Pathfinder/Sming/samples/HttpServer_ConfigNetwork/include/AppSettings.h:34: undefined reference to `ArduinoJson::JsonVariant::asString() const'
out/build/app_app.a(application.o): In function `char const* ArduinoJson::JsonObject::get<char const*>(ArduinoJson::JsonObjectKey) const':
/Users/matous/Pathfinder/Sming/Sming/SmingCore/Network/../../SmingCore/../Services/ArduinoJson/include/ArduinoJson/JsonObject.ipp:23: undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
out/build/app_app.a(application.o): In function `ArduinoJson::TypeTraits::EnableIf<ArduinoJson::TypeTraits::IsSame<char const*, char const*>::value, char const*>::type ArduinoJson::JsonVariant::as<char const*>() const':
/Users/matous/Pathfinder/Sming/Sming/SmingCore/Network/../../SmingCore/../Services/ArduinoJson/include/ArduinoJson/Internals/../JsonVariant.hpp:121: undefined reference to `ArduinoJson::JsonVariant::asString() const'
/Users/matous/Pathfinder/Sming/Sming/SmingCore/Network/../../SmingCore/../Services/ArduinoJson/include/ArduinoJson/Internals/../JsonVariant.hpp:121: undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
out/build/app_app.a(application.o): In function `ApplicationSettingsStorage::load()':
/Users/matous/Pathfinder/Sming/samples/HttpServer_ConfigNetwork/include/AppSettings.h:38: undefined reference to `ArduinoJson::JsonVariant::asString() const'
out/build/app_app.a(application.o): In function `ArduinoJson::TypeTraits::EnableIf<ArduinoJson::JsonObject::CanSet<char const*>::value, ArduinoJson::JsonObjectSubscript<char const*> >::type& ArduinoJson::JsonObjectSubscript<char const*>::operator=<char const*>(char const*)':
/Users/matous/Pathfinder/Sming/samples/HttpServer_ConfigNetwork/app/application.cpp:213: undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
out/build/app_app.a(application.o): In function `bool ArduinoJson::JsonObject::setNodeAt<bool>(ArduinoJson::JsonObjectKey, bool)':
/Users/matous/Pathfinder/Sming/Sming/SmingCore/../Wiring/IPAddress.h:30: undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
out/build/app_app.a(application.o): In function `ArduinoJson::TypeTraits::EnableIf<ArduinoJson::JsonObject::CanSet<String&>::value, ArduinoJson::JsonObjectSubscript<char const*> >::type& ArduinoJson::JsonObjectSubscript<char const*>::operator=<String>(String const&)':
/Users/matous/Pathfinder/Sming/samples/HttpServer_ConfigNetwork/app/application.cpp:135: undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
out/build/app_app.a(application.o): In function `onAjaxConnect(HttpRequest&, HttpResponse&)':
/Users/matous/Pathfinder/Sming/samples/HttpServer_ConfigNetwork/app/application.cpp:147: undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
out/build/app_app.a(application.o):(.text._Z17onAjaxNetworkListR11HttpRequestR12HttpResponse+0x8f): undefined reference to `ArduinoJson::JsonObject::getNodeAt(char const*) const'
out/build/app_app.a(application.o):/Users/matous/Pathfinder/Sming/samples/HttpServer_ConfigNetwork/app/application.cpp:76: more undefined references to `ArduinoJson::JsonObject::getNodeAt(char const*) const' follow
[... cut ...]
/Users/matous/Pathfinder/Sming/Sming/compiler/lib//libsming.a(HttpServer.o): In function `Delegate<void (WebSocket&)>::operator()(WebSocket&) const':
/Users/matous/Pathfinder/Sming/Sming/Wiring/WVector.h:47: undefined reference to `WebSocket::enableCommand()'
/Users/matous/Pathfinder/Sming/Sming/compiler/lib//libsming.a(HttpServer.o): In function `HttpServer::removeWebSocket(HttpServerConnection&)':
/Users/matous/Pathfinder/Sming/Sming/Wiring/WVector.h:47: undefined reference to `WebSocket::~WebSocket()'
collect2: error: ld returned 1 exit status
make: *** [out/build/app.out] Error 1

```

alonewolfx2, 2016-05-07:
propably you are using old version of sdk or sming.

matoushybl, 2016-05-07:
OK, I'll try to update. Is this the correct SDK I should be using: https://bintray.com/kireevco/generic/esp-alt-sdk/view ?

alonewolfx2, 2016-05-07:
kireevco's esp-alt sdk not fully tested on windows and linux but kireevco
also using mac too. so i dont know well about esp-alt-sdk on mac but your
issue coming from linker. so there is a problem with sdk and sming versions

matoushybl, 2016-05-07:
I have tried tu run clean of Sming and it seems to be fixed. Thanks for your help.

closed 2016-05-07 by matoushybl