SmingHub/Sming #728: Problems installing websocketclient

jasperachtbaan, 2016-05-09:
Hello everyone,

I have just installed the websocket client for sming (https://github.com/hrsavla/Sming/tree/feature/websocket-client/Websocket_Client). The websocket client itself works great, but the websocket server demo doesn't work anymore. I get these errors:
`c:/tools/sming/Sming/SmingCore/Boards.h:17: undefined reference to `WebSocket::sendString(String const&)'
C:\tools\sming\Sming/Wiring/WVector.h:47: undefined reference to `WebSocket::~WebSocket()'
:\tools\sming\Sming/Wiring/WVector.h:47: undefined reference to `WebSocket::enableCommand()'`
When I remove the pwm.h, pwm.cpp, websocketclient.h and websocketclient.cpp and rebuild the sming\sming folder everything works fine again. I'm really stuck with this since I'm developing a websocket application where both the client and the server side are needed.

Thanks in advance,

Jasper

hreintke, 2016-05-17:
@jasperachtbaan : 
What I don't understand id the relationship between the pwm and websocket.
When you remove websocketclient, aren't you back to the "standard sming" without updates ? 

BTW : the websocket client is not yet part of sming. There has been unfinished work on that by @hrsalva. 
Probably among others see #202 and #258.
If you finish them off within SmingRTOS, that can be included in there.

jasperachtbaan, 2016-05-17:
I don't understand the relationship between the pwm.h and the websocketclient as well, but when I try to build the websocketclient example without pwm.h and pwm.cpp, it doesn't work. And yes, when I remove the websocketclient it goes back to normal Sming.

Jasper

patrickjahns, 2016-05-17:
@jasperachtbaan 

First thing - did you try building the Sming source from the complete PR by @hrsavla ? Does the error there also occur? 

After that, having a quick look at the sourcecode reveals the dependencies

[WebsocketClient.h](https://github.com/hrsavla/Sming/blob/feature/websocket-client/Sming/SmingCore/Network/WebsocketClient.h)
```
[...]
#include "../PWM.h"
```

Looking into PWM.h  the only includes are:
[PWM.h](https://github.com/hrsavla/Sming/blob/feature/websocket-client/Sming/SmingCore/PWM.h)
```
#include "../Wiring/WiringFrameworkDependencies.h"
#include "../Wiring/WVector.h"
```

Besides the Wiring include nothing relevant is in PWM.h or PWM.cpp

My suggestion:
Remove the unnecessary dependency on pwm.h from WebsocketClient.h and fix the remaining "Wiring" include paths

jasperachtbaan, 2016-05-18:
Hello, thank you for the suggestions. I tried making sming from hrsavla's PR, but I only got a bunch of errors (replaced the Sming folder and build, building went fine, when trying to build any project got errors). Removing the dependencies also gave errors when trying to build the websocketclient (undefined references). It could be that I made a mistake somwhere, but I really wouldn't know where.

Jasper

patrickjahns, 2016-05-18:
I checked the PR again and it`s pre Travis - so it is hard to know if this PR was ever working. Probably only @hrsavla is able to shine light on it.

I`d go with @hreintke suggestion by rewritting the current / writing your own implementation based on/against Sming RTOS

hreintke, 2016-05-23:
Websocket Client implementation should go to SmingRTOS.

closed 2016-05-23 by hreintke