SmingHub/Sming #733: HwPWM glitches

johndoe8967, 2016-05-26:
with the SDK there are several issues with the HwPWM
- can't reach 100%
- period not exact (1,1ms instead of 1ms)
- additional 75us puls at the end

see also: https://goo.gl/bd1asj

but there is also an official response to the issue with some source files to fix the problems
http://bbs.espressif.com/viewtopic.php?f=31&t=1378
might help in future to fix the problem at least in sming

hobby16, 2016-09-20:
Sming "hardware pwm" is indeed unusable because of glitches. "software pwm" is even worse, it's unstable !
Below is my fix to have clean pwm with sming. It's not restricted to 1024 resolution. The resolution is 1/5 microsecond so a very precise pwm is possible, for example for servo signals (for a period of 20ms : HW_pwm.setPeriod(100000); ). 
- delete C:\Espressif\ESP8266_SDK\lib\**libpwm.a**
- replace it with a debugged libpwm.a by compiling it from **pwm.c**

**pwm.c** comes from https://github.com/StefanBruens/ESP8266_new_pwm . See author's site on how interrupt is used to implement clean PWMs.

To compile **pwm.c** into a library **libpwm.a**:

- copy **pwm.c** in C:\Espressif\ESP8266_SDK\lib\
- compile it to obj 
> C:\Espressif\xtensa-lx106-elf\bin\xtensa-lx106-elf-gcc.exe -c pwm.c -o pwm.o  -Os -ggdb -std=gnu99 -Wno-error -Wno-implicit -Wpointer-arith -Wundef -Wall -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -D__ets__ -DICACHE_FLASH -D_STDINT_H -Wno-address -I C:\Espressif\ESP8266_SDK\include

- convert obj (.o) to library (.a)
> C:\Espressif\xtensa-lx106-elf\bin\xtensa-lx106-elf-ar.exe rcs libpwm.a pwm.o

now you have a nice libpwm.a that is a drop-in replacement of Espressif's buggy libpwm.a

P.S. compiling **pwm.c** into **libpwm.a** can be done more nicely with a Makefile but I don't know how to do it. 
I've tried it on 4 output, works great. The example with **8 outputs didn't work** but I have not searched why.

hreintke, 2016-09-20:
@hobby16 : 
Please use SmingRTOS. Th HWPWM there is without the mentioned issues

hobby16, 2016-09-20:
@hreintke 
On SmingRTOS, HWPWm is marqued as 

> compiles, not tested on hardware

On NonOS, simple projects like 'blink' or 'interrupt' compile and run ok. But with more interesting projects like 'Meteo' or websocket, I've got a lots of problem compiling (errors like 'irom0_0_seg' overflowed, etc) and when compiled, I have .BINs that don't start eg 
> load 0x40100000, len 28720, room 16 
> tail 0
> chksum 0xb3
> load 0x3ffe8000, len 892, room 8 
> tail 4
> chksum 0xb8
> load 0x3ffe8380, len 8312, room 4 
> tail 4
> chksum 0xf2
> csum 0xf2
> csum err
> ets_main.c 

Sming is really, really great but personnally, user's experience has been so far disastrous contrary to Arduino sketches. Arduino is crappy (takes ages to compile and upload) but all the projects, and I've tried many of them, work seamlessly. 

I think I will end up using SmingRTOS, only read recently from you that it is prefered over SmingNonOS. 
If only I had known it earlier... There should have been big warning(s) about it in the SmingNonOS page, something like `USE SmingRTOS instead`. 
With SmingRTOS, I'll check if HWPWm is ok and I'll report back. Anyway, I'll stick to my libpwm because I need very good pwm resolution and 1024 steps (worse with servo, only 180 steps !) are not enough.

hobby16, 2016-09-21:
Ok, I've tried HWPWm with SmingRTOS, I can confirm it works great, _without_ glitch.

slaff, 2016-10-16:
> pwm.c comes from https://github.com/StefanBruens/ESP8266_new_pwm
> now you have a nice libpwm.a that is a drop-in replacement of Espressif's buggy libpwm.a

@anakod How about adding this as core part of Sming NONOS? The change does not need to be ported to SmingRTOS because there it is not relevant.

slaff, 2016-11-24:
@johndoe8967 @hobby16 The latest develop version has support for using custom PWM based on Stefan Bruens PWM.

closed 2016-11-24 by slaff