SmingHub/Sming #735: Fresh installed Sming in Windows by Choco:make: esptool2: Command not found

williamesp2015, 2016-06-05:
After automatic installation by Choco in Windows needs esptool2 but no information where to create and how to address in windows PATH. I tried to get it from https://github.com/raburton/esptool2 and compiled but no success. Any help would be appreciated

kennpie, 2016-06-05:
CHERTS' Unofficial Dev Kit includes the source for esptool2 which, after install, can be found at C:\Espressif\examples\ESP8266\rboot\esptool2\. I have MS Visual Studio Comunity edition installed; I just used VS to open the .sln file in that directory, and VS built the esptool2.exe without any problems. Then I moved esptool2.exe to somewhere in my path, and it worked.

williamesp2015, 2016-06-05:
I resolved after struggle with esptool2 finally I copied esptool2 files in the c:\tools\esptool2 then added an Eciipse environment variable (Windows->preference>C/C++ >build>environment name=Esptool2 and value=C:/tools/esptool2/esptool2.exe

hreintke, 2016-06-21:
Sorry for the inconvenience. 
The choco installation & documentation on that is not clear.\

If you have experience with using that, help is appreciated.

closed 2016-06-21 by hreintke