SmingHub/Sming #739: Error with "libmicroc.a" and function "strtok"

CocoRid, 2016-06-07:
Hello,

I try to use Sming to program my ESP8266.

When I compile, I have the following error : 
```
/home/dijouxc/Sming/Sming/compiler/lib//libmicroc.a(strtok.o):(.irom0.literal+0x4): undefined reference to `strspn'
/home/dijouxc/Sming/Sming/compiler/lib//libmicroc.a(strtok.o):(.irom0.literal+0x8): undefined reference to `strcspn'
/home/dijouxc/Sming/Sming/compiler/lib//libmicroc.a(strtok.o): In function `strtok':
C:\DEV\ESP8266\esp_microc/string/strtok.c:7: undefined reference to `strspn'
C:\DEV\ESP8266\esp_microc/string/strtok.c:9: undefined reference to `strcspn'
```

I need function "strtok", but apparently with this function, an error occured ... 
I am on an Ubuntu 14.04.

Can you help me ? Or did I have to find another function to replace "strtok" ?

Regards

hreintke, 2016-06-21:
You can use the function splitString for this.

closed 2016-10-21 by slaff

anakod, 2016-10-24:
Yes, Sming have limited functionality in standard C library and doesn't provide full linux libc implementation (it's too big for ESP). splitString should be good option for you.