SmingHub/Sming #742: HttpServer_AJAX: Connection dropped due to memory?

ajaybhargav, 2016-06-09:
When running this sample application, Everything works fine at start. Webpage loads and I can see continuous GET request for /ajax/input. ESP serves the request untill it reaches a point there is no more heap memory available.
This only happens when heap size drops around 6K (Free heap size=6656, K=0). If I wait a while and try to refresh the page I can see the requests getting served again (heap grows around 11K). At this time if page keeps on sending requests heap again drops to 6K and connection drops. Is there something that memory is not actually freed when tcp connection closes?

Here is debug message for reference:

```
onAccept state: 0 K=0
Free heap size=6656, K=0
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient: 192.168.0.100

path=/ajax/input
Host === 192.168.0.105
parsed
Request: GET, nodata
TCP received: 347 bytes
onReadyToSendData: 1
response sendHeader
response sendBody
realloc 1 -> 131
realloc 130 -> 260
realloc 259 -> 325
realloc 324 -> 390
realloc 389 -> 455
Stream completed
TCP connection closing
~TCP connection
-TCP connection

CONNECTION DROPPED
        (6472)

CONNECTION DROPPED
        (6472)

CONNECTION DROPPED
        (6472)

CONNECTION DROPPED
        (6472)

CONNECTION DROPPED
        (6472)

CONNECTION DROPPED
        (6472)
```

hreintke, 2016-06-21:
This is most certainly due to the (slow) freeing of buffers within LWIP.

I do not see this behavior within SmingRTOS, best to convert to that.

ajaybhargav, 2016-06-21:
ok, RTOS is better.

closed 2016-06-21 by ajaybhargav