SmingHub/Sming #744: HttpServer_WebSocket sample not working

johndoe8967, 2016-06-12:
I had a short look at the HttpServer_WebSocket and there are at least 2 bugs with prevent from easy using
- Makefile-user.mk: spiffs is disabled, should be 
# DISABLE_SPIFFS = 1
SPIFF_FILES = web/build

- web folder: wrong index.html (is a copy bug from the AJAX sample)

hreintke, 2016-06-21:
@johndoe8967 
Did a quick check and indeed not correctly.
If you can make a PR for both SmingNONOS and SmingRTOS I will merge.

johndoe8967, 2016-06-25:
problem seems to be only in  SmingNONOS 
RTOS already uses spiffs

slaff, 2016-12-05:
@johndoe8967 That issue should have been fixed in the latest source code. Feel free to reopen the ticket if the HttpServer_WebSocket is still not working for you.

closed 2016-12-05 by slaff