SmingHub/Sming #745: about adding a "link exception" to the license

rahmanih, 2016-06-14:
Hi,

First of all well done for this nice piece of code!
is it possible to add a "link exception" clause in the license, that we can use Sming into a closed source
application ;)

regards
Haithem.

hreintke, 2016-06-14:
@rahmanih 
I am (co)owner but not the original author and not yet very familiar with licensing.
Will do some reading on this and get back to you shortly.

BTW : Support will move from SmingNONOS to SmingRTOS. Better to use that to be future proof.

BTW2 : Any indication on the application you are building ?

rahmanih, 2016-06-15:
Thanks for the quick answer!

BTW2: I'm integrating the esp8266 with an STM32F7 discovery board to use it for wireless communiction with a PC.

regards.

slaff, 2016-11-24:
@rahmanih Sming's license is LGPL. It means that if you make changes to Sming you must release them under LGPL. But if you only use Sming as a library without modifying it then there is no problem including it in your proprietary application. As long as there is a clear separation between Sming and your application AND you have specified that you are using Sming with link to the github repo then everything should be fine.

closed 2016-11-24 by slaff