SmingHub/Sming #746: Can't flash samples: No such file or directory: 'out/firmware/0x00000.bin'

DominicMe, 2016-06-15:
I installed Sming by following the Linux wiki but there is an issue with either Sming or the Wiki. If I run make command for Basic Blink sample it works fine  but any other sample fails with the following error:

`make: Nothing to be done for 'all'`

make flash fails with the following error:

```
make flash 
stat: cannot stat 'out/firmware/0x09000.bin': No such file or directory
Killing Terminal to free /dev/ttyUSB0
make: [flash] Killed (ignored)
/opt/Sming/Sming/Makefile-project.mk:338: recipe for target 'flash' failed
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0xC000 out/firmware/spiff_rom.bin
usage: esptool write_flash [-h] [--flash_freq {40m,26m,20m,80m}]
                           [--flash_mode {qio,qout,dio,dout}]
                           [--flash_size {4m,2m,8m,16m,32m,16m-c1,32m-c1,32m-c2}]
                           [--no-progress] [--verify]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'out/firmware/0x00000.bin'
make: *** [flash] Error 2
/opt/Sming/Sming/Makefile-project.mk:338: recipe for target 'flash' failed
```

There are indeed no 0x00000.bin or 0x09000.bin files in firmware directory unlike in the BasicBlink example. Why? How do I flash any samples?

fernandomorse, 2016-06-15:
Try : 

make clean
make
sudo make flash

:)

DominicMe, 2016-06-15:
So simple. Why did Blink sample work? Shouldn't "all" run clean as well? Why the "make: Nothing to be done for 'all'" error? Maybe there's a wiki explaining to these options (not sure what to look for)?

Maybe you can chime in on another smaller issue, I get an error complaining about missing esptools2, so I run `export PATH=$PATH:$ESP_HOME/esptool2` but that only works till reboot, any solution to this?
I also don't remember being dropped into miniterm after flashing, is that normal and has it allays been like this?

hreintke, 2016-06-21:
@DominicMe : 
The build process looks for the esptool2 application based on the environment variable ESPTOOL2. Setting that to the correct path will make it work.

On miniterm, that has always been like that. 
Indeed better to be able to avoid. If you create a PR with a generic solution for bot smingNONOS and SmingRTOS I will merge.

DominicMe, 2016-06-21:
So currently the only solution is to run export PATH=$PATH:$ESP_HOME/esptool2 every time after reboot or add it to run automatically on boot?

Can you clarify what you mean by:

> Indeed better to be able to avoid. If you create a PR with a generic solution for bot smingNONOS and SmingRTOS I will merge.

closed 2016-07-24 by DominicMe