SmingHub/Sming #754: Module connects to network without code to do so (even Basic_Blink)

DominicMe, 2016-07-15:
I am having a bizzare issue where no matter what code I upload the module connects to my router. I did not have this issue until recently when I entered the ssid and pass in one of the projects. Now any project I successfully flash gets connected to the network. Basic_blink has no network connectivity code for example but I get the following output:

```
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
Killing Terminal to free /dev/ttyUSB0
make: [/opt/Sming/Sming/Makefile-project.mk:339: flash] Killed (ignored)
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin
esptool.py v1.2-dev
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0000
Writing 28672 @ 0x0... 28672 (100 %)
Wrote 28672 bytes at 0x0 in 2.5 seconds (92.0 kbit/s)...
Writing 196608 @ 0x9000... 196608 (100 %)
Wrote 196608 bytes at 0x9000 in 17.0 seconds (92.3 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�����8���mode : sta(18:fe:34:a1:43:af)
add if0
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with SKY220DB, channel 1
dhcp client start...
ip:192.168.0.63,mask:255.255.255.0,gw:192.168.0.1
pm open,type:2 0

```

I can confirm it is actually connected to the router. I run make clean then make and finally make flsh. Serial output does change depending on project so it is actually a new project that is flashed. Seems like some sort of leftover cache but how can I get rid of it? I though flashing removed everything on the module.

DominicMe, 2016-07-15:
It seems to be related to Eclipse environment variables:
```

// If you want, you can define WiFi settings globally in Eclipse Environment Variables
#ifndef WIFI_SSID
	#define WIFI_SSID "PleaseEnterSSID" // Put you SSID and Password here
	#define WIFI_PWD "PleaseEnterPass"
#endif
```

The problem is that I restored it to the default value on the only sample project I changed it in and still same issue. I restarted the machine with no help. These variables do not appear in Eclipse Environment variables settings page. How am I supposed to reset it?

alonewolfx2, 2016-07-15:
Its Espressif sdk related. You can erase ssid vars with flash init target.

DominicMe, 2016-07-15:
Thanks for your reply. I tried running `make flash init` but it is still connecting to the network. I also tried 4 different modules so it almost seems like it's not anything on the module but rather how it is compiled though it is possible that modules were flashed with the network variables previously.

fvpalha, 2016-07-15:
I have the same problem. The new SDK (1.5.4) does not change the SSID and PASSWORD in STATION mode.

avr39-ripe, 2016-07-15:
Esptessiff SDK stores ssid/password for sta / softap in separate flash sector outside those which you reflash by different firmware.. moreover - it stores in this separate place default wifi mode enabled/disabled combination, so if you enable wifi sta mode, provide ssid and pwd for you net, and indirectly SAVE all this by espressif sdk - next time even with different firmware which do not directly enable wifi sta, nor provide ssid pwd, you module will connect to previously stored net.. you can use sming WifiStation / WifiAccessPoint classes to gain control what to use/enable temporary or store permanently to config area of esp sdk.

DominicMe, 2016-07-15:
What worked for me was `python2 /opt/esp-open-sdk/esptool/esptool.py erase_flash`. I assume that essentially restores the module to factory defaults?

@avr39-ripe Are you saying that having for example `WifiStation.enable(true);` in a project and flashing would disable for any other firmware that doesn't explicitly set it? Flashing a new firmware with different ssid and pass didn't seem to fix it for me.

avr39-ripe, 2016-07-15:
WifiStation.enable have SECOND optional argument that if true stores sta enable by default in esp sdk config area, so if you flash another fw and do NOT explicitly WifiStation.enable(false) or even WifiStation.enable(false, true) wifi sta mode will be enabled by default with prev stored ssid/pwd. you can also use WifiStation.config's optional arg to store or not ssid/pwd in esp sdk config area

DominicMe, 2016-07-15:
Ok, I see it now. Very annoying behavior but at least now I know how to factory restore it :). Thanks for the explanation.

closed 2016-07-15 by DominicMe