SmingHub/Sming #756: WifiStation isn't firing Event

gikvin, 2016-07-15:
I have 2 event set on WifiStation connection and fail. First time it is firing one of them correctly but
after losing and then reconnecting to wifi, wifi station isn't firing connection event!

[image]

and no event after reconnecting wifi.

avr39-ripe, 2016-07-15:
how you set wifi events? do you use WifiEvents? if so what events do yo catch by you callback / delegate?  what do you mean lost connection? is wifi router turn off/on or how?

gikvin, 2016-07-16:
I'm using this method
`WifiStation.waitConnection(connectOk, 30, connectFail);`
I'm not using WifiEvents.
If wifi router goes on and off i want my connectOk function to be called again.

hreintke, 2016-07-16:
WifiStation.waitconnection is designed and implemented to fire only once.
if you need more control/information of Wifi status you should use WifiEvents.

gikvin, 2016-07-16:
Thank you!
I think this issue needs to be closed.

closed 2016-07-16 by hreintke

gikvin, 2016-07-16:
Meago,yw y