SmingHub/Sming #757: FTP Fatal exception 9(LoadStoreAlignmentCause):

DominicMe, 2016-07-16:
I am trying to get ftp working but every time I attempt to connect I get an error followed by wdt reset. I tried using ftp example only adding minimal WifiStation code and still get the error. What could be the cause? I also tried WifiAccessPoint with similar results though different error.

Relevant parts of the application.cpp file:

```
void startServers(){
	debugf("DEBUG: start servers");
	startWebServer();
	startFTP();
	//WifiStation.connect();
}

void init(){
        spiffs_mount();
        WifiStation.enable(true, false);
        WifiStation.config(AppSettings.defaultSsid, AppSettings.defaultPassword, false);
        WifiStation.waitConnection(onStationConnect, 20, onStationFail);
}

```
Serial output from flashing completion to reset:

```
formated
mount res: 0

mode : sta(18:fe:34:a4:35:41) + softAP(1a:fe:34:a4:35:41)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
DEBUG: start servers
'/' registered
'/config' registered
'/config-save' registered
'/ajax/get-networks' registered
'/ajax/connect' registered
'/api/sensors' registered
addUser: me 123
add 1
aid 1
station: 68:17:29:b6:dc:0d join, AID = 1
onAccept state: 0 K=0
Free heap size=37472, K=0
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient: 192.168.4.2

path=/index.html
Name: host
host === 192.168.4.1
parsed
Request: GET, nodata
TCP received: 423 bytes
onReadyToSendData: 1
Default server handler for: '/index.html'
SEND ERROR PAGE
TCP connection closing
~TCP connection
-TCP connection
onAccept state: 0 K=0
Free heap size=37304, K=0
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient: 192.168.4.2

path=/
Name: host
host === 192.168.4.1
parsed
Request: GET, nodata
TCP received: 387 bytes
onReadyToSendData: 1
open errno -10002

File wasn't found: index.html
SEND ERROR PAGE
TCP connection closing
~TCP connection
-TCP connection
onAccept state: 0 K=0
Free heap size=37120, K=0
+TCP connection
timeout updating: 70 -> 900
TcpServer onClient: 192.168.4.2

AUTH: 'TLS'
> 530 FAIL

AUTH: 'SSL'
> 530 FAIL

USER: 'me'
> 331 OK

PASS: '123'
checkUser: me 123
> 230 OK

PWD: ''
> 257 "/"

TYPE: 'I'
> 250 OK

PASV: ''
> 500 Passive mode not supported

PORT: '192,168,4,2,218,181'
connection to: 192.168.4.2, 55989
> 200 OK

LIST: ''
+TCP connection
TcpConnection::connect result:, 0
> 150 Connecting

OnConnected
timeout updating: 70 -> 300
TCP connected
send file list: 0
TCP connection closing
~TCP connection
> 226 Transfer Complete.

Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x4023f958, epc2=0x00000000, epc3=0x00000000, excvaddr=0x20363262, depc=0x00000000

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

vsirvent, 2016-10-22:
Same here. 
I have SPIFF with 6 files I can access using Web server so I discard that spiff filesystems is failing.

Log trace:
`PASV: ''
> 500 Passive mode not supported
PORT: '192,168,4,2,168,191'
connection to: 192.168.4.2, 43199
> 200 OK
LIST: ''
+TCP connection
TcpConnection::connect result:, 0
> 150 Connecting
OnConnected
timeout updating: 70 -> 300
TCP connected
PHMeter::onMeasure: adc = 456
send file list: 6
TCP connection closing
~TCP connection
> 226 Transfer Complete.
TCP sent: 308
Fatal exception 28(LoadProhibitedCause):
epc1=0x40248652, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000a2, depc=0x00000000
 ets Jan  8 2013,rst cause:2, boot mode:(1,6)
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)
wdt reset
`

vsirvent, 2016-10-22:
Now working with WifiStation enabled. It seems that could have something to do with the TcpServer when WifiStation is disabled. I have no idea right now I'll try to debug it...

hreintke, 2016-10-23:
@vsirvent : 
Can you check in which function it breaks using the generated map file & epc1 address ?

flexiti, 2016-11-25:
Similarly:
SPIFF_SIZE      ?= 524288
 SPIFF with 6 files (HttpServer_ConfigNetwork example)
 plus 1 text file added by these demo
FTP connection:

```
connection to: 192.168.0.60, 61376
> 200 OK

LIST: ''
+TCP connection
TcpConnection::connect result:, 0
> 150 Connecting

OnConnected
timeout updating: 70 -> 300
TCP connected
send file list: 7
TCP connection closing
~TCP connection
> 226 Transfer Complete.

TCP sent: 358
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x4023be1a, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0a0d2ed7, depc=0x00000000

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 1588, room 16 
tail 4
chksum 0xb6
load 0x3ffe8000, len 700, room 4 
tail 8
chksum 0xfe
csum 0xfe
```

flexiti, 2016-11-25:
```
New compilation and file maps of this compilation:

Fatal exception 9(LoadStoreAlignmentCause):
epc1=**0x4023be3a,** epc2=0x00000000, epc3=0x00000000, excvaddr=0x0a0d2ed7, depc=0x00000000
```
.text._ZN13TcpConnection6onSentEt
                0x4023be18       0x3d c:/tools/sming/Sming/compiler/lib/\libsming.a(TcpConnection.o)
                                 0x41 (size before relaxing)
                **0x4023be1c                TcpConnection::onSent(unsigned short)**
 *fill*         0x4023be55        0x3 
 .text._ZN13TcpConnection9onReceiveEP4pbuf
                0x4023be58       0x50 c:/tools/sming/Sming/compiler/lib/\libsming.a(TcpConnection.o)
                                 0x58 (size before relaxing)
                0x4023be60                TcpConnection::onReceive(pbuf*)

STwilight, 2016-11-27:
After manual updating Sming core to last release version (2.1.5) I also got this exception (Fatal exception 9(LoadStoreAlignmentCause)) on standard FTP server example with no changes in it's code.
The same code works perfect on 2.1.0 Sming core version.
Seems other things (like HTTP Server), works well on 2.1.5.

slaff, 2016-12-06:
@DominicMe The current version has WDT fix for the FTPserver issues. Please, check and if it is not working for you, feel free to reopen the ticket.

closed 2016-12-06 by slaff

flexiti, 2016-12-07:
@slaff 
Can You show me which patch pls ?

slaff, 2016-12-07:
@flexiti This one 19d977a6a1267e7429ddd9c9eedf6266149e3b48.

flexiti, 2016-12-07:
Thanks @slaff 
ps 
Every time when we use WDT refresh, it means that we pauses "internal" functions of ESP which can cause future errors related with eg. TCP, timers etc.
Engineers ESP expressly cautioned not to occupy the CPU for more than 500 ms in one event. 
If we need to refresh the WDT,it means that we occupy processor for a few seconds.
(internal functions implemented by the manufacturer do not have time to act, among other things, do not have time to refresh WDT)
Any need for refresh WDT **is a sign that our program is faulty**