SmingHub/Sming #758: Serial port parity is not configurable

vortex314, 2016-07-19:
I'm developing a ESP8266 driver that is able to program a STM32 by using the STM32 bootloader. This protocol requires 8bit-even parity. In the actual HardwareSerial this is not configurable.
Any shortcut to do this anyway ?

slaff, 2016-10-31:
@vortex314 It is possible to set the parity manually. For example you can use a function like this:

```C
void
UART_SetParity(UART_Port uart_no, UART_ParityMode Parity_mode)
{
    CLEAR_PERI_REG_MASK(UART_CONF0(uart_no), UART_PARITY | UART_PARITY_EN);

    if (Parity_mode == USART_Parity_None) {
    } else {
        SET_PERI_REG_MASK(UART_CONF0(uart_no), Parity_mode | UART_PARITY_EN);
    }
}
```
For details look in your sdk  folder for `driver_lib/driver/uart.c` and `sdk/driver_lib/include/driver/uart.h`.

slaff, 2017-02-09:
@vortex314 The latest `develop` version has refactored HardwareSerial that allows setting the parity https://github.com/SmingHub/Sming/blob/develop/Sming/SmingCore/HardwareSerial.h#L110 . Feel free to reopen the issue if that does not work for you.

closed 2017-02-09 by slaff