SmingHub/Sming #759: How to send binary data array from server to UART?

r44083, 2016-07-28:
I am using ```myTcpClient.send(buff, buff_length, false)``` for sending requests to the server and callback ```nmOnReceive(TcpClient& client, char *buf, int size)``` for receiving data from server. My callback looks like this:
```
bool nmOnReceive(TcpClient& client, char *buf, int size)
{
	Serial.write(buf);
}
```

But there is a problem when I try to **receive binary data**. This method ```Serial.write(buf);``` isn't suitable for binary data, because in binary data can be ```\r```, ```\n``` or ```\0```.

Does Sming have function which allows to send data buffer with its length to the UART?

So, how can we solve this?

hreintke, 2016-07-28:
Will see for a better solution but for now you should be able to : 
```
for (int i=0;i<size;i++) {Serial.write(buf[i]);
```

slaff, 2016-10-21:
@r44083 Fixed with that commit https://github.com/SmingHub/Sming/commit/c58e8fca806935b73cda2f54fa05661757c1b475

closed 2016-10-21 by slaff