SmingHub/Sming #760: DHT_Humidity sample fails with wdt reset

DominicMe, 2016-07-28:
I am using ESP8266-07 (SMD) module with a 3 wire 2301 sensor. The problem is no matter the configuration it always fails and resets. Enabling debugging in DHT library doesn't not add any more output.

Serial output directly after flashing:
```
Read using Adafruit API methods
Failed to read from DHT

Read using new API methods
Failed to read from DHT: 2Heatindex: nan*C
Dewpoint: nan*C
Comfort is at nan percent, (Unknown:1073645454)
mode : sta(18:fe:34:a4:35:41) + softAP(1a:fe:34:a4:35:41)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100

 ets Jan  8 2013,rst cause:4, boot mode:(1,0)

wdt reset
```

What I tried:

- With and without 10k resistor between data pin and 3.3v.
- Different pins - 2, 4, 14
- External 3.3v power
- For DHTTYPE constant I tried AM2301, DHT22 and omitting it for the default value.

The code is the same as the DHT_Humidity sample with only the following lines modified:
```
#define WORK_PIN 2
#define DHTTYPE AM2301

DHT dht(WORK_PIN, DHTTYPE);
```

I tried the same AM2301 sensor with same breadboard connections with Arduino UNO and it worked fine on pin 2 as DHT22 type. It did not work on pin 4 but didn't reset either.

What is going wrong here?

si9, 2016-08-08:
Try this simple example code:
```
#define WORK_PIN 0 // GPIO0 
DHT dht(WORK_PIN, DHT22);
Timer procTimer;
TempAndHumidity th;
void readTH()
{
 if ( ! dht.readTempAndHumidity(th))
  {
    Serial.println("Failed to read from DHT");
  }
  else
  {
    Serial.print("H: ");
    Serial.print(th.humid);
    Serial.print("% T1: ");
    Serial.print(th.temp);
    Serial.print(" *C");
  }
}
void init()
{
  dht.begin();
  procTimer.initializeMs(1000, readTH).start();
 }
```

slaff, 2016-10-31:
@DominicMe can you report the current status of issue? Are the WDT resets still there?

DominicMe, 2016-10-31:
@slaff I haven't tried, will have to give it a try again. I assume there was an update, was it the sample of the framework or both?

slaff, 2017-01-11:
@DominicMe Feel free to reopen the issue if the problem is still happening for you.

closed 2017-01-11 by slaff