SmingHub/Sming #762: rc-switch: crash if receiver connected

667bdrm, 2016-08-04:
The controler crashes if the program uses receive functionality and RF receiver connected (tried XY-MK-5V and RXB6). The program loads correct if disconnect receiver before the boot and connect later. Also program crashes on access to any defined page url. Here is a debug output:

load 0x40100000, len 24524, room 16 
tail 12
chksum 0x90
ho 0 tail 12 room 4
load 0x3ffe8000, len 888, room 12 
tail 12
chksum 0x14
ho 0 tail 12 room 4
load 0x3ffe8380, len 10244, room 12 
tail 8
chksum 0xc1
csum 0xc1
rl�+TCP connection
timeout updating: 70 -> 65535
Server timeout updating: 40 -> 90
+TCP connection
timeout updating: 70 -> 65535
Server timeout updating: 40 -> 900
+TCP connection
fs.start: size:192 Kb, offset:0x4c000

mount res: 0

Fatal exception 0(IllegalInstructionCause):
epc1=0x40244268, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

hreintke, 2016-08-04:
@667bdrm : 
Maybe you wanted to send this to another list but here and with the information above there is no way to understand your issue and provide help to solve it.

si9, 2016-08-08:
I had a similar problem when I can not compile spiffy spifs file. Perhaps you have the same case.

hreintke, 2016-08-14:
@si9 : 
The mount res : 0 
Is the debug output for a "successful SPIFFS mount"

closed 2016-08-14 by hreintke