SmingHub/Sming #764: Recieving UDP connections disabled?

geokon-gh, 2016-08-14:
I'm just getting started with the ESP8266, so sorry if I'm missing something. 

I'm trying to do a very simple setup for the time being. Read in UDP packets that contain RGB value and output them using the NeoPixel library. I have something similar set up on an Arduino where I get the packet buffer and copy the pixel values over to the NeoPixel LED buffer - but I can't seem to figure out how to do the equivalent with Sming.

There is a UdpConnection class which has methods for sending packets. However the onReceive() method  - which has a packet buffer in its interface - is for some reason virtual and I can't figure out how I'm expected to use it. 

Do I create a child class that extends UdpConnection and implements the onRecieve virtual class? This will then automatically get called (though some kind of automatic call back) when a  the ESP recieves a UDP packet at the port set with the listen() method?

hreintke, 2016-08-14:
When answering this I realized that the UDP is not very intuitive.  Maybe we need an example.
For now : Sming is fully based on Delegates, a special type of callback. This Delegate can be both a class function or an global function.

That is also the reason why there is no loop() within Sming.

For UDP & global function you can use the following : 
The myReceive delegate will be call by Sming on all received packets.
```
void myReceive(UdpConnection& connection, char *data , int size, IPAddress remoteIP, uint16_t remotePort)
{
	connection.sendString("This is the reply");
}
UdpConnection udp(myReceive);
void init()
{
	udp.listen(5001);
}
```

You can look at the Basic_Delegates example on how to use Delegates with a Class Function

geokon-gh, 2016-08-28:
Thanks hreintke,
Thanks for the quick feedback and sorry for the slow response. I got the UDP connection work and I'm now working on the LED part (working off of the example provided). There there is the Timer Delegate which acts similarly (instead of triggering the callback when a UDP packet is recieved, I guess it's triggered when the timer runs down to zero).

What I  can't find is what's the difference between a callback and Delegate?

hreintke, 2016-08-28:
@geokon-gh : 
The Timer delegate is can be used in "repeating mode" and "once mode", triggering repeating or once after defined interval.

The Delegate is special against the callback in that it can be both a "global function" as a "class function".
F.e in the basic_blink you see the use of a global function in Timer
in Basic_Delegate you see the use of a class function.

All "callbacks" within Sming are implemented as Delegates.

geokon-gh, 2016-09-06:
Thank you for all the help

closed 2016-09-06 by geokon-gh