SmingHub/Sming #765: Equivalent functions of yield() or delay() in SMING

helpme1, 2016-08-17:
ESP8266 Arduino has yield(), delay() to move processing to the CPU when certain processes are hogging CPU resources for too long. Does SMING have equivalent functions for yield() and delay()? If no, it would be a good feature to add.

Does SMING's delay() achieve the same effect as Arduino's delay() for moving processing to the CPU?

hreintke, 2016-08-17:
@helpme1 : 
SmingRTOS, which is based on the Espressif RTOS SDK, has the RTOS as it's base.
Because of the inherent multitasking features of this, there is no need for the yield() and adapted delay().
The RTOS kernel will handle the regular processing of the SDK (runs in a separate task).
The WDT reset is prevented in this way.

SmingRTOS is the port from SmingNONOS and has already more functionality.
SmingRTOS is the one which will be supported.

helpme1, 2016-08-17:
@hreintke:
Thanks for the reply. For the time being, if I need a quick fix with my SmingNONOS app, are there convenient functions to use for something like yield()?

I do not know how tedious it is to migrate from SmingNONOS to SmingRTOS. Any quick guide on migration?

closed 2017-01-20 by slaff

BotoX, 2017-12-20:
Could this be reopened?
Kinda sucks not having yield(), also SmingRTOS being dead.