SmingHub/Sming #770: CTS/RTS implemented?

vsirvent, 2016-09-19:
Are CTS/RTS pins for serial comm implemented? Should I go down to the esp8266 api?

Thanks!

Vicente

crosofg, 2016-10-07:
yes they are implemented I guess. Always worked for me.

slaff, 2016-10-16:
I do not understand exactly what you mean?

Sming can control the UART communication and the ESP8266 pins.
The UART communication usually has two cables RX and TX. If you are talking about the CTS/RTS pins in an USB cable, where Sming has NO control, this is a completely different story. Some ESP8266 boards, like MCU for example expose the CTS and RTS pins for that firmware uploaders like `/esptool.py` put the device into flash mode and restart it afterwards without the need of changing jumpers. 

Sming itself is a software library. And as such it does not control peripheral which is outside of its control.

closed 2016-10-24 by slaff