SmingHub/Sming #772: rboot - serial flash after OTA

ido4pro, 2016-09-22:
The command 'make flash'  upload rom in slot 0.

But, depends of the last OTA, the rboot could start in slot 1.

How force the board to boot again to the slot 0 by the serial port when OTA is broken ?

Thanks for your help,

someburner, 2016-09-22:
See `BOOT_CUSTOM_DEFAULT_CONFIG` in rboot.h. Set that, recompile, erase the rboot sectors, then flash.

ido4pro, 2016-09-22:
Thanks, I fix my issue.

BOOT_CUSTOM_DEFAULT_CONFIG in rboot.h is only available in the original project : https://github.com/raburton/rboot/ -> Not today, in the sming framework

I add line 367 of rboot.c : romconf->current_rom = 0;

And i increment the BOOT_CONFIG_VERSION in rboot.h to force the boot to enter to this portion of code

```
// fresh install or old version?
	if (romconf->magic != BOOT_CONFIG_MAGIC || romconf->version != BOOT_CONFIG_VERSION
#ifdef BOOT_CONFIG_CHKSUM
		|| romconf->chksum != calc_chksum((uint8*)romconf, (uint8*)&romconf->chksum)
#endif
		) {
		// create a default config for a standard 2 rom setup
		ets_printf("Writing default boot config.\r\n");
		ets_memset(romconf, 0x00, sizeof(rboot_config));
		romconf->magic = BOOT_CONFIG_MAGIC;
		romconf->version = BOOT_CONFIG_VERSION;
		romconf->count = 2;
		romconf->current_rom = 0;
		romconf->roms[0] = SECTOR_SIZE * (BOOT_CONFIG_SECTOR + 1);
		romconf->roms[1] = (flashsize / 2) + (SECTOR_SIZE * (BOOT_CONFIG_SECTOR + 1));
#ifdef BOOT_CONFIG_CHKSUM
		romconf->chksum = calc_chksum((uint8*)romconf, (uint8*)&romconf->chksum);
#endif
		// write new config sector
		SPIEraseSector(BOOT_CONFIG_SECTOR);
		SPIWrite(BOOT_CONFIG_SECTOR * SECTOR_SIZE, buffer, SECTOR_SIZE);
	}
```

closed 2016-09-22 by ido4pro

raburton, 2016-09-22:
Erase / replace the rBoot config sector (sector 1).