SmingHub/Sming #773: Problem in installing sming on windows!

MustafaFayez, 2016-09-26:
I tried installing sming on windows according to wiki page in this repo, i always get hung in installing esp8266_udk step and the terminal is hung , what should i do?
note: i use windows 7 64bit  and i have installed CHERTS SDK:
https://github.com/CHERTS/esp8266-devkit

is this conflicting with the sming installation?
thanks.

STwilight, 2016-12-28:
@MustafaFayez, try to unnistall version of CHERTS SDK, that you installed, and install 2.0.9 version: https://github.com/CHERTS/esp8266-devkit/releases/tag/v2.0.9.

At this moment only 1.5.4 Espressif SDK is supported (not 2.0.0, that appears in CHERTS SDK from version 2.1.0 and above).

For details see #820.

slaff, 2017-02-09:
SDK 2.0 is also supported in the latest `develop` version. After the release(v3.1.0) at the end of the week(12.Feb.2017) the latest code should be available also Windows/chocolatery users.

closed 2017-02-09 by slaff