SmingHub/Sming #775: Problem with make sming on mac

namlunoy, 2016-10-01:
I'm using Mac OS 10.12
When I run "make" command from /opt/Sming/Sming folder.
I got this error:

"Kim:Sming hoangcong$ sudo make
AS system/irq_check.s
make: /opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: Malformed Mach-o file
make: *** [out/build/system/irq_check.o] Error 1"

Any ideas?

guettomusick, 2016-10-04:
I have the same problem, I try several things and cant fix it. It happened after updating to Sierra or Xcode 8, on El Capitan it works great. Could you solve it?

lucas-bremond, 2016-10-09:
Possibly an issue coming from the `Install prebuilt ESP Alt SDK` step, with `esp-alt-sdk-v1.5.0.258-macos-x86_64.zip` being not compatible with macOS Sierra anymore.

I could get rid of that error by compiling `esp-open-sdk` from source following these instructions:
http://www.esp8266.com/wiki/doku.php?id=setup-osx-compiler-esp8266

Be aware that your default Python has to be 2.7, I had Python 3 as my system wide default and encountered issues. Reverting to 2.7 solved it.

slaff, 2016-10-31:
@namlunoy Can you report the latest status of the issue on you Mac?

harry-boe, 2016-11-22:
Ok here are my findings ..  This is for macOS 10.12 Sierra. 
Before that downloading the rebuild toolchain works just fine.

The process as described by @lucas-bremond is outdated. I did that exact same thong more than a year back.

It will likely report missing help2man packages etc ..
Also the SED patching is no longer required.

What worked for me was strictly following the description on github for the package here:
[https://github.com/pfalcon/esp-open-sdk#macos](https://github.com/pfalcon/esp-open-sdk#macos)

I also noticed that the case sensitive file system is only required for building the toolchain.
Once build it can be copied over to /opt/esp-open-sdk and everything runs fine

slaff, 2017-02-09:
Closing due to inactivity.

closed 2017-02-09 by slaff

OrWest, 2017-04-18:
I had same error on MacOS. 
@harry-boe post has helped me resolve this issue. Thank you.

But additionally, I have removed sdk link in esp-open-sdk folder, because make hadn't want start.
Then make file had finished successfully.