SmingHub/Sming #776: error: no matching function call to 'HttpClient::downloadString(const char[65], <unresolved overloaded function type>'

fernandomorse, 2016-10-14:
I am trying to create a class "IOT" with code below:

File: iot.h
======

#ifndef iot_h 
#define iot_h

class IOT 
{ 
        public: 
            String HttpEnvia();
       private: 
             void CallBack(HttpClient& client, bool successful); 
             HttpClient Http;
};
#endif

file: iot.cpp
=======

#include 
#include "iot.h" 

void IOT::CallBack(HttpClient& client, bool successful) 
{
         if(successful) 
        {
             String resposta = client.getResponseString();
             String codigo = String(client.getResponseCode());
             String body = client.getPostBody();

             Serial.println(resposta);<br>
            Serial.println(codigo);<br>
            Serial.println(body);<br>
         }
        return;
}

void IOT::HttpEnvia() 
{
          if(Http.isProcessing())
          {
                    // nothing for while
           }
           else
           {
                 Http.setRequestHeader("Authorization", "Basic XXXXXXXXXXXXXXXXXXXXX==");
                 Http.setRequestHeader("Connection","Close");
                 Http.setRequestContentType("application/json");            Http.downloadString("http://10.1.1.22/api/device/C13BD5642798FE93B7B6FD23/resource/58",CallBack);
         }
        return;
}

Error after compile: error: no matching function call to 'HttpClient::downloadString(const char[65], <unresolved overloaded function type>'

I dont know fix the error. Any example or idea to fix ??

hreintke, 2016-10-16:
@fernandomorse : 
You are using a class function as callback -> need to specify that in the downloadString call.

CallBack there should be : 
HttpClientCompletedDelegate(&IOT::CallBack,this)
That specifies the function IOT::CallBack and the object this to be used.

fernandomorse, 2016-10-16:
Thanks, i will change the code. tx

fernandomorse, 2016-10-16:
Another question on the same subject:

System.onReady(SystemReadyDelegate(&IOT::Callback,IOT)), dont work. what is my mistake in this line of code. Tks.

File: system.h
Class SystemClass
Delegate => SystemReadyDelegate

hreintke, 2016-10-16:
@fernandomorse : 
The second argument of a delegate should be : A pointer to the object which is handling the callback.
The IOT in your code is the class, not the object.

fernandomorse, 2016-10-17:
@hreintke ,

Sorry my questions. I'm new in C ++.
In code is implemented as follows:

application.cpp

System.onReady(SystemReadyDelegate(&IOT::ServerWEB,*???));

server.cpp

void IOT::ServerWEB()
{
    // stuff
}

server.h

class IOT
{
     public
              ServerWEB();
};

*???? => Pointer to the object. I do not know what I should enter here

Thanks

avr39-ripe, 2016-10-17:
@fernandomorse 
In Sming there are plenty places where you can put some async handler function. Asyncronity and event-based model is strong part of Sming in contrast to Arduino for esp8266. So to bussiness - each time you whant provide some event-handler you should provide either some callback-function or Delegate. difference between them is that plain callback function is simple function accessible in part of code where you put it as event handler:

```C++
void tickHandler();
Timer tickTimer;
tickTimer.initializeMs(1000, tickHandler);
tickTimer.start(true); //arm tickTimer to start tickHandler after 1000 ms AND then restart this timer again and again (true) argument to start() if .start(false) run timer only ONCE!
void tickHandler()
{
  Serial.println("TICK!");
}
```
So here tickHandler is simple function attached to timer event.
Now Delegate.. If you use C++ Classes and whant some class-member function to act as some event handler you must use Delegate concept which binds Class-method to some class-instance and use it as event handler. You cannot use non-static class member by itself without instance context.
```C++
Class Ticker {
public:
 void tickHandler();
}

void Ticker::tickHandler()
{
  Serial.println("TICK!");
}

Ticker myTicker;

Timer tickTimer;
tickTimer.initializeMs(1000, TimerDelegate(&Ticker::tickHandler, myTicker));
tickTimer.start(true); 
```
so here TimerDelegate(&Ticker::tickHandler, myTicker) binds class-method Ticker::tickHandler to instance myTicker, so you can use it as timer event handler.
Rather simple description, but I think sufficient-enough.
Delegates is another superb part of Sming framework :)

fernandomorse, 2016-10-17:
Great @avr39-ripe ,

Thanks , examples opened my mind.

closed 2016-10-21 by slaff