SmingHub/Sming #777: Spi library slave

Matheus-Garbelini, 2016-10-19:
Hello, Could it be possible to port SPI library so esp could work in master or slave mode?

harry-boe, 2016-12-05:
The SPI implementation supports only master mode right now. 
Slave mode is a completely different story as you need to handle interrupts when traffic comes.
If you like to do it the espressif sdk (examples/peripherial_test) would provide a starting point.

What is the setup where you need slave mode?

Matheus-Garbelini, 2016-12-17:
@harry-boe I need a simples setup like what esp8266 core SPISlave currently have:
https://github.com/esp8266/Arduino/tree/master/libraries/SPISlave

It trigger an interrupt Avery 32bytes (i guess). If im not mistaken, me-no-dev configure the spi using the DMA to handle the data transfer. Please, Look forward to this kind of implementation on sming as it would certainly increase sming Framework to work with other very fast systems. (Like raspberry-esp)

closed 2017-12-19 by Matheus-Garbelini