SmingHub/Sming #783: NONOS & RTOS

hreintke, 2016-10-24:
@slaff @anakod 
See both of you being active again on Sming, especially SmingNONOS.

In my opinion we should not get into the situation that there is a SmingNONOS & SmingRTOS with different functionality. 
What are your plans on this ?

slaff, 2016-10-24:
@hreintke If there are people that are interested and want to contribute I would not stop them. If they want to contribute to Sming NONOS it should be fine, as long as the changes are ported to RTOS. This way both projects will profit. 

Now, my personal opinion, is that the NON OS SDK is a bit more stable, advanced and predictable compared to the FreeRTOS SDK. Again, this is my own opinion. And for my sort-of-commercial project I need something that is stable as much as it can get. So I see Sming NONOS as the right choice for ESP8266.

On the other hand ESP32 is out. And it uses `esp-idf` which is FreeRTOS based. And here SmingRTOS should have been the right choice. I wrote "should" because SmingRTOS is a step in the right direction to make it work under ESP32 but it still needs a lot of work to run with the `esp-idf`. Again, my personal opinion is, that SmingRTOS should start moving to  `esp-idf` if we want to make Sming work also there.

slaff, 2016-10-27:
> I cannot/will not/want to be the only one supporting SmingRTOS.

@hreintke You have the unique opportunity to be the team leader of the SmingRTOS project. As a start just merge pull requests that are ported from Sming and this will show that the SmingRTOS project has more than one contributor.

hreintke, 2016-10-27:
@slaff @anakod 
Me being the one merging pull requests from NONOS into SmingRTOS is the one thing I am avoiding.
That is also not getting more contributors to SmingRTOS, it is having someone trying to keep it more or less in sync.  Exactly why I stated "I cannot/will not/want to be the only one supporting SmingRTOS."

If you/anakod/community want to focus on SmingNONOS, no problem from my side.
I then will try to keep SmingRTOS as it is, updating when I see major issues, add functionality to already added features and merge PR's submitted to SmingRTOS.

slaff, 2016-10-28:
@hreintke @anakod My idea is to finish all current PRs. After that PRs adding new drivers will be redirected to SmingRTOS. PRs improving the existing Sming code will be welcomed and migrate to RTOS (as long as they are relevant there).

hreintke, 2016-11-25:
@slaff @anakod : 
Seeing the recent deviations between SimgNONOS and SminRTOS I do not feel it is feasible to keep them in sync. 
Because of that, I will stop supporting/adding functionality to the SmingRTOS repository. 
For my applications I started the migration to Arduino-ESP8266 with AsyncTCP/AsyncWebServer. That will provides a similar async network framework as Sming and a common codebase for my applications on ESP8266 & ESP32 (using Arduino-ESP32).

slaff, 2016-11-27:
@hreintke I am sad to hear that :( I would like to thank you for your help, valuable advices and source code contributions. The party is [just starting](https://github.com/SmingHub/SmingRTOS/issues/140) so make sure to be around :)

closed 2017-03-11 by slaff