SmingHub/Sming #785: Support to Mongoose Flash Tool (fastest flashing time)

Matheus-Garbelini, 2016-10-26:
Hello, Could you guys support in the makefile the Mongoose FlashTool?  It has CLI and It is an optimized flash tool that allows upload speed faster then 1000000 baud rate depending on the ttl serial to usb converter in the board. It also optimize writing, which in total allow upload time to around just **5 seconds**.

Here's the github: [https://github.com/cesanta/mongoose-flashing-tool](url)
And Here's the blog describing another optimization they made with the normal esptool: [https://blog.cesanta.com/esp8266-superfast-flashing-flash-esp-in-5-seconds](url)

Matheus-Garbelini, 2016-10-26:
@slaff @hreintke anyone?

slaff, 2016-10-27:
@Matheus-Garbelini In a way the Cesenta / Mangoose flasher is already part of the [esptool](https://github.com/themadinventor/esptool) that Sming uses for flashing. 

> upload speed faster then 1000000 baud
As long as your hardware supports that baudrate this is true for both the pure Cesenta flasher and esptool.

hreintke, 2016-10-27:
@slaff @Matheus-Garbelini 
Nice functionality is the autodetect of flash size. 
Did not look into any details on how to integrate.

slaff, 2016-10-28:
> Nice functionality is the autodetect of flash size. 

@hreintke That is present in esptool since the latest versions. If there is a huge win in using the `mongoose-flashing-tool` and it does not require big refactoring may be we can add it to SmingRTOS.

Matheus-Garbelini, 2016-10-29:
@slaff Thank you a lot. Actually when you said about using the normal esptool in python I didn't understood, because, as i'm using your sming branch (the most updated in my opinion) I can just use esptool2, if I change the makefile configuration to esptool.py path will it work?

closed 2016-12-06 by slaff