SmingHub/Sming #786: Basic_WebSkeletonApp Fatal exception (28)

itarataiko, 2016-10-28:
Hi,
I'm trying to run Basic_WebSkeletonApp.
After flashing code I see next exception:

Killing Terminal to free /dev/tty.usbserial
/Volumes/case-sensitive/esp-open-sdk//esptool/esptool.py -p /dev/tty.usbserial -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0x48000 out/firmware/spiff_rom.bin
esptool.py v1.2-dev
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0000
Writing 40960 @ 0x0... 40960 (100 %)
Wrote 40960 bytes at 0x0 in 3.6 seconds (92.2 kbit/s)...
Writing 253952 @ 0x9000... 253952 (100 %)
Wrote 253952 bytes at 0x9000 in 22.0 seconds (92.3 kbit/s)...
Writing 196608 @ 0x48000... 196608 (100 %)
Wrote 196608 bytes at 0x48000 in 17.0 seconds (92.3 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/tty.usbserial 115200
--- Miniterm on /dev/tty.usbserial  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
28):
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
Fatal exception (28):
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
Fatal exception (28):
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
Fatal exception (28):
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
Fatal exception (28):
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000

itarataiko, 2016-10-28:
I'm using this board for prototyping http://www.elecrow.com/esp8266-iot-board-p-1591.html

itarataiko, 2016-10-29:
after commenting startWebServer(); in the init() app starts without exception

can anybody help me?

avr39-ripe, 2016-10-29:
There is some bugs in Sming core networking. I will point you bit later where fix them, from mobile so very restricted..

itarataiko, 2016-10-29:
Thanks for your answer! I'll wait for the fix

avr39-ripe, 2016-10-29:
Hmm.. can't reproduce your issue.. Checkout recent Sming/develop, rebuild sming, rebuild Basic_WebSkeletonApp, make flashinit, make flash and it just works.. As after flashinit there is no saved configuration, so app just starts own AP with no auth.. then you can connect to device webui and enter ssid/passwd of you wifi network. then it connects to you wifi network in STA mode and shuts down own AP.

avr39-ripe, 2016-10-29:
Also look at my PR #668 and #669

itarataiko, 2016-10-29:
thanks for your help.
I'll try once more on Monday morning and write about my results.

avr39-ripe, 2016-10-29:
Ah, SmingNONOS.. I test it under SmingNONOS.

itarataiko, 2016-10-29:
hmmm.. and how will I test this sample?

avr39-ripe, 2016-10-29:
What is the problem to test? There are two SDK for eso8266 - one RTOS-based and one nonos.. So here are two Sming versions, simple Sming or SmingNONOS (it is THE SAME, just different name!!) and SmingRTOS. I said that i test app useing Sming = SmingNONOS, not SmingRTOS.. :)

itarataiko, 2016-10-29:
I tested it with simple Sming too yesterday :)
ok, I'll test it once more and write about results

Matheus-Garbelini, 2016-10-29:
hey @itarataiko how did you managed to use esptool.py with sming with its makefile? I Could only use esptool2 that is supported and documented in sming. There's some place in sming that tells how to configure it to the makefile automatically use python esptool? I didn't find anithing. Thanks

MartinHerren, 2016-10-30:
I try to migrate a project to use Sming, using the Basic_WebSkeletonApp as a template, and I have the same issue since a few weeks. On development and master branch, just tested it again with today's dev branch:

    Fatal exception (28): 
    epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000

If i understand correctly it's a memory access violation, code at 0x4000228b tries to access the illegal address 0x000000b5.
Still if i understand correctly the offending code is at flash's address 0x0000228b, which according to my .map file corresponds to some microc.a code.

In fact beside the Basic_Blink example, all example i tried show this behavior. I'm using a wemos.cc LoLin dev board.

If i comment out

    spiffs_mount();

and change the http code to only serve

    response.forbidden();

i don't have this problem and have a nice web server capable of responding 403s only, which led me to think it's a flash issue. But even trying to serve static strings using

    response.sendString("Test");

leads to the same error, so i'm quite confused now.

avr39-ripe, 2016-10-30:
I use recent SmingNONOS develop branch + espressif sdk 1.5.4 + esp-alt-sdk
as gcc 5.x compiller under linux/freebsd.. and all seems to work.

itarataiko, 2016-10-30:
@MartinHerren under which OS are your try to compile this app?

itarataiko, 2016-10-30:
I try to compile under MacOS.

MartinHerren, 2016-10-30:
For my part under Linux.

Did some more testing. I followed the quick start guide from https://github.com/SmingHub/Sming/wiki/Linux-Quickstart so i was using the latest SmingNonOS along with the latest ESP-OpenSDK.

As without Sming my project used an older OpenSDK i tried the same with Sming and it worked. Pulling commit from january this year based on SDK 1.4.0 works. Current SDK defaults to version 2.0.0

So the problem is probably based on the sdk version used.

Will test some more to identify the latest working sdk. Perhaps the quickstart should be updated not to pull latest esp-open-sdk.

MartinHerren, 2016-10-31:
So latest git pull of esp-open-sdk works if you replace

    VENDOR_SDK = 2.0.0

with

    VENDOR_SDK = 1.5.4

in the top level Makefile.

@itarataiko if you used latest esp-open-sdk too, try that, recompile esp-open-sdk, Sming and your code and check again.

itarataiko, 2016-10-31:
@MartinHerren thanks for your help! after changing VENDOR_SDK = 2.0.0 with VENDOR_SDK = 1.5.4 and recompiling all sdk's my code working well!

once more Thanks for help!

closed 2016-10-31 by slaff