SmingHub/Sming #788: Delegate attachInterrupt

fernandomorse, 2016-10-31:
I need an example for code: attachInterrupt (GPIO, interrupt CONFIG, LOW); using delegate.

hreintke, 2016-11-02:
@fernandomorse : 
There is a basic_delegates example which shows the generic use of delegates.

Noticed that we missed the tyedef in the attachInterrupt,.

For attacheInterrupt it you can use would be like 
```
typedef Delegate<void()> InterruptDelegate;
class example 
{
 void interruptRoutine()
{...}
...
}
```
In the class code you can then use : 
```
attachInterrupt(GPIO,InterruptDelegate(&example::interruptRoutine,this),LOW) 

```

fernandomorse, 2016-11-02:
Grateful, realized changes in interrupt.h. Working perfectly.

Please, close topic.

closed 2016-11-03 by hreintke