SmingHub/Sming #792: RCSwitch.send() causes wdt reset

juppin, 2016-11-03:
Hello developers,

the Library RCSwitch is not fully optimzed for Sming.
Mostly all calls in my project to the send(code, codeLength) function of the Library RCSwitch causes a reset through wdt.
I have tried to disable the wdt with WDT.enable(false) and system_soft_wdt_stop(), but my esp8266 is still reseted by wdt.

I think the delays in the Library causes the reset.

I have also tried to set the repeatTransmit to 1.

Hence i will use a dedicated Arduino Nano, connected over i2c, to transmit my codes.
So this issue is low priority.

slaff, 2016-12-06:
@anakod Can you take a look at that issue?

slaff, 2016-12-07:
@juppin Can you can get the latest version of the source code, completely rebuild Sming and your application? When it gets reseted by WDT it will dump stack information. Decoded that information and paste the decoded information here.

slaff, 2016-12-29:
@juppin With the additions like crash handler in the latest Sming code you should be able to point us to the  watch dog reset issue. Can you get the latest source code, compile it and if the WDT reset is present to show us the stack dump.

juppin, 2017-01-02:
I will take a look at this that week. Yesterday i have newly installed Sming Framework on my Windows maschine with the "Windows Quickstart" guide with Chocolatey. I have tested the Basic_Blink and the HttpServer_AJAX examples.

I see that Sming Framework version is on 2.1.0. Is that the latest source that i should use for?
If not, how can i get the newest Framework version on Windows with or without Chocolatey?

I don`t want to use my old vagrant box, i think that can cause more errors...
Otherwise i will setup a new Ubuntu VM.

juppin, 2017-01-07:
This issue could be closed.

My application is now working as it should.

I think the RCSwitch example should also work.

Thank you for your work on Sming. I love it.

closed 2017-01-08 by slaff