SmingHub/Sming #799: Preparation for Release 2.1.5

slaff, 2016-11-14:
This issue is created to collect all the tasks we need to do before the 2.1.5 Release.

### Source Code
- [ ] Update Sming/SmingCore/SmingCore.h 
- [ ] Update Readme.md
- [ ] Tag the source code

### Pending PRs
- [X] Merge Docker add-ins to the Wiki documentation 
- [ ] Redundant externs #798

### Statistics

- [ ] We should collect information about the RAM/Flash usage for this release.

### Needs testing support/feedback from the community

* Docker 
* API docs
* rBoot
* printf with "fancy" formatting
* Overall performance and issues

MartinHerren, 2016-11-14:
I successfully built my app using the provided docker. I can't get the same binaries (beside the spiffs) as using my own container even if i use the same sdk and sming versions. But as my base image is debian 8 with gcc 4.9 instead of 4.8 that could explain that the generated toolchain diverges somewhat.
But
 - building
 - flashing
 - running

the firmware works with the provided container. Openend PR #801 to freeze a specific esp-open-sdk commit to be able to obtain the same toolchain at a later time again.

Will try to check some of the Sming samples if they build and run fine.

Will try the C9 interface as well.

slaff, 2016-11-21:
Basic Blink statistics:
```
#Memory / Section info:
------------------------------------------------------------------------------

out/build/app.out:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000375  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       000002ec  3ffe8380  3ffe8380  00000460  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00006370  3ffe8670  3ffe8670  00000750  2**4
                  ALLOC
  3 .irom0.text   0002f300  40209000  40209000  00006840  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         000060eb  40100000  40100000  0000074c  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
------------------------------------------------------------------------------
```

closed 2016-11-21 by slaff