SmingHub/Sming #800: Preparation for Release 2.2.0

slaff, 2016-11-14:
This issue is created to collect all the tasks we need to do before the 2.2.0 Release.

### Source Code
- [ ] Update Sming/SmingCore/SmingCore.h 
- [ ] Update Readme.md
- [ ] Tag the source code

### Pending PRs
- [ ] Better structure and usage of third-party repositories #793
- [ ] SSL support, using axTLS, restructured so that it comes as third-party repository

### SmingRTOS
- [ ] Reapplying PR #793 to SmingRTOS so that we minimize the effort and files that need to be changed/migrated.

### Statistics

- [ ] We should collect information about the RAM/Flash usage for this release.

### Needs testing support/feedback from the community

* Overall performance and issues

slaff, 2016-12-04:
@alonewolfx2 @robotiko Can you test the Basic_Neopixel sample from the latest source code and tell me if it is working on your devices?

harry-boe, 2016-12-05:
i still struggle with the toolchain after the OSX Sierra update.
Maybe someone can point me to some instruction on how to use the Docker container to get around the OSX issues ?

slaff, 2016-12-05:
> how to use the Docker container

@harry-boe Read the documentation: https://github.com/SmingHub/Sming/wiki/Docker-Quickstart.

slaff, 2016-12-07:
@anakod Can you check if the Radio_RCSwitch sample is working on a real device with the latest code? It would be great if you can help us test other samples that have unknown status in the list above.

closed 2016-12-24 by slaff