SmingHub/Sming #803: Write Firmware Sming ESP8266 model Esp-01

fernandomorse, 2016-11-15:
I'm trying to write the Sming firmware on the Esp8266 model esp-01. I made the modification in the Makefile-user.mk file:

## Configure flash parameters (for ESP12-E and other new boards):
SPI_MODE = dio
SPI_SIZE = 1024

and

## Configure flash parameters (for ESP12-E and other new boards):
SPI_MODE = dio
SPI_SIZE = 512

Under both conditions it presents the following error:

```
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm dio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0x4C000 out/firmware/spiff_rom.bin
Connecting...
Erasing flash...
Wrote 34816 bytes at 0x00000000 in 3.4 seconds (81.5 kbit/s)...
Erasing flash...
Writing at 0x0002a800... (52 %) 
Traceback (most recent call last):
  File "/opt/esp-open-sdk/esptool/esptool.py", line 599, in <module>
    esp.flash_block(block, seq)
  File "/opt/esp-open-sdk/esptool/esptool.py", line 221, in flash_block
    raise Exception('Failed to write to target Flash')
Exception: Failed to write to target Flash
/opt/Sming/Sming/Makefile-project.mk:340: recipe for target 'flash' failed
make: *** [flash] Error 1
```

What am I doing wrong, what should I do ??

Link Model Esp-01 url  http://www.filipeflop.com/pd-1f55ad-modulo-wifi-esp8266-esp-01.html

slaff, 2016-11-16:
```python
 raise Exception('Failed to write to target Flash')
```
Most probably you do not have write access to /dev/ttyUSB0. Either try to flash as `root` user or add your current user to the `dialout` group. It can also be that the device is not in FLASH mode.

hreintke, 2016-11-16:
Know in my case it is windows but...

I get this message when another program (putty,...) is using the port on which the esp is connected.

flexiti, 2016-11-16:
or ESP is not in programming mode seting by pin

fernandomorse, 2016-11-16:
Let's step by step:
My user is already part of the dialot group, even when running as root, the error happens;

There is no program accessing the port that the esp is connected, I do not use putty;

Yes, Esp is in the firmware recording mode, I'm using FTDI232, with the following ESP x FTDI pinouts:

ESP FTDI
TX RX
RX TX
VCC VCC
CHPD VCC
GPIO0 GND (Enable Firmware Recording Mode)
GND GND

Using the program of "nodemcu" (esp8266flash), it saves the file, but when running it presents errors.

I've used at least 3 different ESP-01s.

Grateful.
Morse

flexiti, 2016-11-16:
Windows or Linux ?
You are using esp-open-sdk, I'm not familiar with this.
I am using windows and espressif (esptool.exe) and rboot:

> c/Espressif/utils/esptool.exe -p COM8 -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/rboot.bin 0x02000 out/firmware/rom0.bin

and all is ok

fernandomorse, 2016-11-16:
Dear flexiti
I'm using esp-open-sdk on linux platform.

slaff, 2016-12-06:
@fernandomorse Feel free to reopen the ticket is you have more information related to this issue.

closed 2016-12-06 by slaff