SmingHub/Sming #807: Serial.readString() cause en exception if called reapeatedly

icodk, 2016-11-21:
I use Serial.readString() to empty the serial receive buffer (is there a better way of doing it ?).
Happen to call it by mistake two consecutive times and got en exception.
`Serial.readString() ;
Serial.readString() ;`
Calling only two times might succeed, to be sure to get en exception add one more  call to Serial.readString() ;
I am using SDK 1.5.3 and sming  2.1.0

hreintke, 2016-11-21:
@icodk : 
Don't know what exception you get but readString() goes into a busy loop when characters are still needed when buffer is empty, Can cause WDT reset.

icodk, 2016-11-21:
Might be a wdt time out.  It takes just over 2 seconds to reboot 
msg I get is:
`ts Jan  8 2013,rst cause:2, boot mode:(3,6)`

slaff, 2016-11-22:
@icodk The default read timeout is 1 second.  The software watchdog resets after roughly 2 seconds.
If you call twice Serial.readString() without resetting the software watchdog you will reach the 2 seconds and the WDT will reset the application. What you can do is something like this 
```C

Serial.setTimeout(500); // set the timeout to 500 ms...
while(true) {
    WDT.alive();
    String data = Serial.readString();
    if(!data.length()) {
        break;
    }
}
```

> is there a better way of doing it 

That is a good question. The following *might* work and should be blazingly fast:

```C
// UartDev is defined and initialized in ROM code.
extern UartDevice UartDev;

RcvMsgBuff &rxBuf = UartDev.rcv_buff;
// Set pointer to next data word in ring buffer
rxBuf.pReadPos = rxBuf.pRcvMsgBuff ;
```

icodk, 2016-11-22:
@slaff 
Thanks, will try it out. 
May be should be added instead of the empty void HardwareSerial::flush() ?

slaff, 2016-11-23:
@anakod Are you familiar with emptying the UART 0 receive buffer? Can you add a function to HardwareSerial that empties the receive buffer?

hreintke, 2016-11-23:
@slaff : 
Received uart chars are placed into RcvMsgBuff &rxBuf = UartDev.rcv_buff;
setting both rxBuf.pReadPos and pWritePos to rxBuff->pRcvMsgBuff empties the buffer.
On flush() : I would expect that to flush the output buffer, not the inputbuffer.

slaff, 2016-11-23:
>  both rxBuf.pReadPos and pWritePos to rxBuff->pRcvMsgBuff empties the buffer.... On flush() : I would expect that to flush the output buffer, not the inputbuffer.

@hreintke Thanks. And I completely agree with you about flush. Is SmingRTOs having already a method that does this? If not can you suggest some name that is "consistent" with the Arduino naming?

hreintke, 2016-11-23:
No there is no option for that in SmingRTOS.
AFAIK there is no common Arduino name.
Be aware to that the result of emptying can be very short lived. There is even no guarantee that the buffer is actually empty at return of the function. A new char can already be arrived just before the actual return. 

Emptying the buffer in application code can easily be done by  : 
`While Serial.availbable() { Serial.read()};`

Other option is to fully disable the use of the internal HardwareSerial by using a callback with useSerialRxBuffer = false. Then the application is responsible for using/buffering received chars.

icodk, 2016-11-29:
There could be use cases where it is beneficial to clear the buffer. For example: a system with a keypad without a display, or half duplex communication to another system. Anyway we now have a way to do it.
so we can close this question from my side.

closed 2016-11-30 by slaff