SmingHub/Sming #817: Basic_Ssl sample failing to compile

riban-bw, 2016-11-27:
The basic ssl sample project is failing to compile. make is complaining about multiple definition of 'millis'.

slaff, 2016-11-27:
@riban-bw Thanks for reporting the issue exactly in the form above. It gave me a very good idea for a series of FAQs. The first one, based on your report, will be: 

** Sample X does not compile! What shall I do  **
The first thing that you need to do is to make sure that you have a clean source code. 
And second if the sample is still not compiling you have to provide us with more information. 

Let's start with the first part: "Clean Source Code State".  If you are familiar with `git` you can run `git status` to get more information. Sometimes this won't be enough therefore we recommend you the following steps ( They are working on Linux with bash shell. If you have another OS and shell you should adjust them accordingly). 

```bash
cd /tmp 
git clone https://github.com/SmingHub/Sming.git SmingForTest
cd /tmp/SmingForTest/Sming
export SMING_HOME=/tmp/SmingForTest/Sming
```

The commands above will fetch the latest source code in a directory that should be completely different than the Sming directory that you are using on a daily basis. Also it points the SMING_HOME variable to the new temporary directory with the clean source code.

Now let's go to the second step: "Compile a sample and report issues, if any". We will use the Basic_Ssl sample. Before we compile a sample we need to compile the Sming library. This can be done calling the following commands:
 
```bash
cd /tmp/SmingForTest/Sming
export SMING_HOME=/tmp/SmingForTest/Sming
make clean
```

The last makes sure to clean any intermediate files or directories. If we run now `make`. It will fetch the needed submodules, compile the code and build a library out of it. In our case we need to compile Sming with an optional SSL support. In order to compile Sming with SSL we need to add the ENABLE_SSL=1 directive.  The command that we need to run now will look like this:

``` bash
make ENABLE_SSL=1 
``` 

**On error**
If the compilation stops with an error, please, copy the output that the command above produces and append it to your bug report. Now run the same command one more time but with the V=1 directive. This will give produce a more verbose output that can help the Sming developers figure out the issue. 

```
make ENABLE_SSL=1 V=1 
```

Make sure to append that output too to your bug report. Tell the Sming developers also what is your operating system, version, git version, make version so that the developers can reproduce your problem. 

**On success**
It is time to compile the Basic_Ssl sample. Do this by executing the commands below:

```bash
cd /tmp/SmingForTest/Sming/samples/Basic_Ssl
export SMING_HOME=/tmp/SmingForTest/Sming
make clean
make
```

** On error 2 **
If that compilation fails make sure to append the output to your bug report. Now compile the sample with the V=1 flags, similar to the compilation of the Sming library.

```bash
cd /tmp/SmingForTest/Sming/samples/Basic_Ssl
export SMING_HOME=/tmp/SmingForTest/Sming
make V=1 
```

And ( I am sure you already know it but let me repeat it again ) append this output to the bug report.

Thank you for your help. And please, remember: The more useful information you provide the easier it will be to find a solution for the problem.

Your Sming Team :)

riban-bw, 2016-11-27:
Thanks Slaff. First, note that your instruction to
`cd /tmp/SmingForTest/Sming/samples/Basic_Ssl`
should actually say
`cd /tmp/SmingForTest/samples/Basic_Ssl`

Output from make within Basic_Ssl directory:

``C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out
/home/brian/src/esp-open-sdk/sdk/lib/libmain.a(time.o): In function `millis':
(.irom0.text+0x94): multiple definition of `millis'
/tmp/SmingForTest/Sming/compiler/lib//libsmingssl.a(time.o):/tmp/SmingForTest/Sming/third-party/axtls-8266/replacements/time.c:33: first defined here
/home/brian/src/esp-open-sdk/sdk/lib/libmain.a(time.o): In function `micros':
(.irom0.text+0xd0): multiple definition of `micros'
/tmp/SmingForTest/Sming/compiler/lib//libsmingssl.a(time.o):/tmp/SmingForTest/Sming/third-party/axtls-8266/replacements/time.c:30: first defined here
/home/brian/src/esp-open-sdk/sdk/lib/libmain.a(time.o): In function `configTime':
(.irom0.text+0x138): multiple definition of `configTime'
/tmp/SmingForTest/Sming/compiler/lib//libsmingssl.a(time.o):/tmp/SmingForTest/Sming/third-party/axtls-8266/replacements/time.c:34: first defined here
/home/brian/src/esp-open-sdk/sdk/lib/libmain.a(time.o): In function `time':
(.irom0.text+0x1dc): multiple definition of `time'
/tmp/SmingForTest/Sming/compiler/lib//libsmingssl.a(time.o):/tmp/SmingForTest/Sming/third-party/axtls-8266/replacements/time.c:30: first defined here
collect2: error: ld returned 1 exit status
/tmp/SmingForTest/Sming/Makefile-project.mk:306: recipe for target 'out/build/app.out' failed
make: *** [out/build/app.out] Error 1``

Output from make V=1:

``/home/brian/src/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -L/tmp/SmingForTest/Sming/compiler/lib/ -L/home/brian/src/esp-open-sdk/sdk/lib -T/tmp/SmingForTest/Sming/compiler/ld/eagle.app.v6.cpp.ld -nostdlib -u call_user_start -Wl,-static -Wl,--gc-sections -Wl,-Map=out/firmware/firmware.map -Wl,--start-group -lmicroc -lmicrogcc -lhal -lphy -lpp -lnet80211 -llwip -lwpa -lmain -lsmingssl -lcrypto -lpwm -lsmartconfig -laxtls out/build/app_app.a -Wl,--end-group -o out/build/app.out
/home/brian/src/esp-open-sdk/sdk/lib/libmain.a(time.o): In function `millis':
(.irom0.text+0x94): multiple definition of `millis'
/tmp/SmingForTest/Sming/compiler/lib//libsmingssl.a(time.o):/tmp/SmingForTest/Sming/third-party/axtls-8266/replacements/time.c:33: first defined here
/home/brian/src/esp-open-sdk/sdk/lib/libmain.a(time.o): In function `micros':
(.irom0.text+0xd0): multiple definition of `micros'
/tmp/SmingForTest/Sming/compiler/lib//libsmingssl.a(time.o):/tmp/SmingForTest/Sming/third-party/axtls-8266/replacements/time.c:30: first defined here
/home/brian/src/esp-open-sdk/sdk/lib/libmain.a(time.o): In function `configTime':
(.irom0.text+0x138): multiple definition of `configTime'
/tmp/SmingForTest/Sming/compiler/lib//libsmingssl.a(time.o):/tmp/SmingForTest/Sming/third-party/axtls-8266/replacements/time.c:34: first defined here
/home/brian/src/esp-open-sdk/sdk/lib/libmain.a(time.o): In function `time':
(.irom0.text+0x1dc): multiple definition of `time'
/tmp/SmingForTest/Sming/compiler/lib//libsmingssl.a(time.o):/tmp/SmingForTest/Sming/third-party/axtls-8266/replacements/time.c:30: first defined here
collect2: error: ld returned 1 exit status
/tmp/SmingForTest/Sming/Makefile-project.mk:306: recipe for target 'out/build/app.out' failed
make: *** [out/build/app.out] Error 1``

slaff, 2016-11-28:
@riban-bw I can see that some functions are already defined in the SDK that you are using. I can also see that this is esp-open-sdk. Can you send me the version of the SDK and *"...operating system & version, git version, make & version, so that the developers can reproduce your problem..."* ?

riban-bw, 2016-11-28:
SDK version: ESP8266_NONOS_SDK_V2.0.0_16_08_10.
Operating system: Ubuntu Server 16.04 x86 32-bit running within VirtualBox.
(This also happens on a Raspbian Jessie Lite installation.)
git version: 2,7,4.
GNU make version: 4.1.

I followed the installation instructions on the Sming wiki here: https://github.com/SmingHub/Sming/wiki/Linux-Quickstart

$SDK_BASE is not set in my login environment. I believe this is set by makefiles.

Output of  ar -t $ESP_HOME/sdk/lib/libmain.a:
app_main.o
eagle_lib.o
eagle_lwip_if.o
ets_timer.o
mem_manager.o
spi_flash.o
spi_overlap.o
time.o
user_interface.o
debug-vector.o
double-vector.o
kernel-vector.o
nmi-vector.o
user-vector.o
vector.o
user_rf_cal_sector_set.o

slaff, 2016-11-28:
> SDK version: ESP8266_NONOS_SDK_V2.0.0_16_08_10.

@riban-bw Version 2.0.0 is not supported. I can see also that this version comes with `time.o` inside of the library archive libmain.a.  You should use SDK 1.5.x or 1.4.x.

In the [Linux-Quickstart](https://github.com/SmingHub/Sming/wiki/Linux-Quickstart) replace the command `make STANDALONE=y` with `make VENDOR_SDK=1.5.4 STANDALONE=y`.

riban-bw, 2016-11-28:
Okay - that fixed it. I have updated the wiki to match.

closed 2016-11-28 by riban-bw